AndunHH/spacemouse #4: debugOutput3 shows all value to 0

coliss86, 2024-05-30:
Hi,

On my mouse, when moving it nothing happen.

With `debug` set to 3, the output is set to 0:
```
AX:0,AY:0,BX:0,BY:0,CX:0,CY:0,DX:0,DY:0
```
but when setting the debug output to 2, all values are updated : 
```
AX:0,AY:0,BX:0,BY:-241,CX:0,CY:0,DX:0,DY:112
```

After the v3, a new instruction has been added that seems to cause this issue: 
```
 for (int i = 0; i < 8; i++) {
    if (centered[i] < DEADZONE && centered[i] > -DEADZONE) {
      centered[i] = 0;
    }
    else {
      centered[i] = map(centered[i], minVals[i], maxVals[i], -350, 350);  <== this line
    }
```

When commenting this bloc, the mouse is working :)

I think this is incorrect because the upper and lower limits are used to constrain the values ​​sent to the software, not those read by the sensor. This constraint should be applied after all the calculation are done.

WDYT ?

AndunHH, 2024-05-31:
My understanding:
1. First, you try to get the ADC values and assign them correctly. (See our other ticket with the pin assignment, which needs to be fixed)
2. The values are centered in the middle position. Then, with the dead zone some small digits (3 worked for me) in the middle are forced to zero.
3. Now, the interesting part: as some joystick movements use the full range of motion, they go from -500 to e.g. +450. In order to have a equal distribution of this values over the whole range of motion, we find the actual min and max values that occur and map them to the range between -350 to +350. Now all further calculations can be done with this normalized values between +/-350.
4. We calculate the translation and rotation based on this.
5. Applying the modifiers to minimize very small rotations or translations.

I changed, where the modifiers are applied, from the centered data, to the resulting trans and rot data, as this worked way better for me.
In the version by LivingTheDream, the calculations (in 4.) also only calculated some movements and forced the other values to zero. I find the modifiers much better, as they only minimize other movements, but don't ignore them.
(Somehow ike the "dominant" mode in the settings by 3dconnexion, that I don't like)

I will test the debug3 output again and report back.

AndunHH, 2024-05-31:
I created a little semi-automatic script to get the values for the minVals and maxVals. You start this procedure with debug=20.

coliss86, 2024-05-31:
the debug=20 is really great :-) 

I tried again with the current main version and here is the logs while doing a complete RY and RX:
```
AX:0,AY:0,BX:0,BY:0,CX:0,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:0,BY:0,CX:0,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:0,BY:0,CX:58,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:0,BY:0,CX:233,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:0,BY:0,CX:360,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:0,BY:0,CX:462,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:0,BY:0,CX:498,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:0,BY:0,CX:498,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:0,BY:0,CX:498,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:0,BY:0,CX:498,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:0,BX:93,BY:0,CX:498,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:10,BX:205,BY:0,CX:498,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:88,BX:301,BY:0,CX:498,CY:-21,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:115,BX:342,BY:0,CX:498,CY:-81,DX:-5,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:144,BX:323,BY:0,CX:396,CY:-80,DX:-243,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:137,BX:237,BY:0,CX:82,CY:-80,DX:-408,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:108,BX:206,BY:0,CX:0,CY:-75,DX:-525,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:97,BX:201,BY:0,CX:0,CY:-68,DX:-525,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:89,BX:201,BY:0,CX:0,CY:-81,DX:-525,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:0,AY:85,BX:204,BY:0,CX:0,CY:-86,DX:-525,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:341,AY:7,BX:340,BY:0,CX:0,CY:-65,DX:-243,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:499,AY:0,BX:381,BY:0,CX:0,CY:-59,DX:-163,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:500,AY:0,BX:380,BY:0,CX:0,CY:-61,DX:-130,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:500,AY:0,BX:343,BY:0,CX:0,CY:0,DX:-79,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:500,AY:0,BX:207,BY:0,CX:0,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:500,AY:0,BX:0,BY:0,CX:0,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
AX:500,AY:0,BX:0,BY:0,CX:0,CY:0,DX:0,DY:0||TX:0,TY:0,TZ:0,RX:0,RY:0,RZ:0
```
here is the minVals and maxVals:
```
int minVals[8] = {-524, -410, -493, -495, -56, -501, -524, -529};
int maxVals[8] = {499, 500, 530, 528, 498, 521, 28, 493};
```
and sensitivity:
```
float rotX_sensitivity = 1;
float rotY_sensitivity = 1;
```

as you can see, the values from sensors are reported but the rotation is still equals to `0`.

AndunHH, 2024-05-31:
Your minVals CX is very small with -56 and your maxVal for DX with 28 is also very limited. You should confirm with debug=2 that you can not get a greater reading when moving the mouse. 

I updated the script to map: 
- the negative values between minVal and 0 to the range from -350 to 0.
- the positive values between 0 and maxVal to the range from 0 to +350. 
This should improve general behavior, but doesn't explain your constant zeros...

Bus this shouldn't be the problem...
Which modFunc do you use? With modFunc=4 my values are also very often near to zero. I suggest modFunc = 3, but for debugging please put it to modFunc=0.

coliss86, 2024-06-03:
Thanks for your reply maybe one of my joystick is broken, I'll have to find another one and make more tests.

closed 2024-06-03 by coliss86