AndunHH/spacemouse #5: PINLIST is wrong

coliss86, 2024-05-30:
Hi,

There is a mistake in the code about pin assignment to joysticks. 
The order in the code is:
```
int PINLIST[8] = { // The positions of the reads
  A0, // X-axis A
  A1, // Y-axis A
  A2, // X-axis B
  A3, // Y-axis B
  A6, // X-axis C
  A7, // Y-axis C
  A8, // X-axis D
  A9  // Y-axis D
};
```

but according to the wiring diagram from TeachingTech, it is:
```
int PINLIST[8] = {
    // The positions of the reads
    A1, // X-axis A
    A0, // Y-axis A
    A3, // X-axis B
    A2, // Y-axis B
    A7, // X-axis C
    A6, // Y-axis C
    A9, // X-axis D
    A8  // Y-axis D
};
```

it also explains with my mouse is not working :)

AndunHH, 2024-05-31:
I just commited a change to put the pin definitions into a config.h file so everyone can have there on pin settings. And we don't overwrite the pin mapping or calibration, when changing other stuff in the code. You just need to copy and rename the config_sample.h to config.h.

Please check, if this resolves this issue.

coliss86, 2024-05-31:
Fixed with the new `config.h`

closed 2024-05-31 by coliss86