AndunHH/spacemouse #13: Reverse Direction and Speed options in software not working

voxpop9, 2024-06-15:
As per the title, i have tried to reverse and slow down the mouse using the 3Dconnexion's software and it is not working.

AndunHH, 2024-06-15:
Yes, this is a known problem now. The 3dconnexion driver doesn't save this setting on the pc, but probably send this information to the spacemouse. And our version doesn't recognize this settings.

I would also like to have this feature! We need to find a solution. Maybe some other project has already solved this issue and we can merge the solution? I would start by looking into the history of the projects, where teaching tech sourced the HID interface.

voxpop9, 2024-06-16:
> Yes, this is a known problem now. The 3dconnexion driver doesn't save this setting on the pc, but probably send this information to the spacemouse. And our version doesn't recognize this settings. [...]

That's interesting, because the Arduino code itself mentions using the software to invert it. perhaps 3Dconnexions changed this at one point.

AndunHH, 2024-06-16:
You are right, but I am sure, this was not working on windows ...

For Linux this is a different story, as the original driver is from 2014 and spacenavd is a popular alternative, which allows this. I linked it in the readme, if you are interested.

voxpop9, 2024-06-16:
> You are right, but I am sure, this was not working on windows ...
> 
> For Linux this is a different story, as the original driver is from 2014 and spacenavd is a popular alternative, which allows this. I linked it in the readme, if you are interested.

Im on Windows so it foesn't apply to me unfortunately. Oh well, i only use it for Onshape at the moment so code level modifications will suffice for now.

coliss86, 2024-06-30:
I don't know how to implement this, the USB bus should be sniffed with a genuine mouse and the protocol should be reversed engineered. A lot of work...

AndunHH, 2024-07-21:
To let you know:
I did an overhaul of the USB HID interface and I'm not done yet -  in the branch [EnableLED](https://github.com/AndunHH/spacemouse/tree/EnableLED)

There are many reports in the HID descriptor that are not used now. I found out that report ID
1. IN Translation
2. IN Rotation
3. IN Keys
4. OUT LEDs
5. not existent
6. OUT Usage (0x21)
7. OUT Usage (0x22)
8. OUT Usage (0x23)
9. OUT Usage (0xB0)
10. OUT Usage (0x25)
11. OUT Usage (0x26)
...

'19. OUT Usage (0x2E)

See https://github.com/AndunHH/spacemouse/blob/main/SpaceNavigator.md#hid-report-of-space-navigator for the whole HID descriptor.

I guess, that those six reports will tell the spacemouse about the sensitivities for the single axis. I have no clue, what report 19 will tell us ;) ... After finishing the LEDs, this will be next and is recorded here as a reminder.

AndunHH, 2024-08-28:
My last comment is missing some important information: the observed hid descriptor with the additional reports are observed in a classical space navigator, which is NOT equal to our emulated space mouse pro!

I tried to switch our emulation to such a space navigator to get the configuration going. Unfortunately, this failed, because the driver didn't send any additional information.???  As expected the keys stopped working because the classic space navigator has only two keys. Therefore we stay we the actual emulation of the space mouse pro, where at least all keys are working.

If someone has such a space mouse pro, so we can Wireshark the USB communication, this would help!

Mr-Falat, 2025-01-28:
in my opinion this settings parameters send to Spacemouse to another endpoint or send by custom feature defined in HID Descriptor ID-6 to ID-19 !
bc when I made any change to any parameters on Setting Window I can see RX LED of my Arduino Pro micro is light up ! that is show data has send from software to arduino.

AndunHH, 2025-01-28:
Hi

I also thought about that.

I got hands on a classical **space navigator** (the old mouse with only two buttons) and observed the communication with Wireshark. The capture can be found here:

https://github.com/AndunHH/spacemouse/blob/main/Reverse-Engineering-Docs%2FSpaceNavigator-Connected-Calibrate.pcapng

The hid descriptord are also extracted and linked in the post above.

Unfortunately, when I pressed calibrate on my PC the only communication was to toggle the LED on and off to show some blinking. I couldn't see anything else on other endpoints.

That's why help is needed by someone to Wireshark the communication of a real **space mouse pro.**

Mr-Falat, 2025-02-06:
Hi everybody .
I found something about atmega32u4 endpoints that may help :

https://www.avrfreaks.net/s/topic/a5C3l000000UcghEAC/t162507

can anybody check this link ?

AndunHH, 2025-02-06:
> in my opinion this settings parameters send to Spacemouse to another endpoint or send by custom feature defined in HID Descriptor ID-6 to ID-19 ! bc when I made any change to any parameters on [...]

Have you checked, if only the LEDs are toggled at this moments or are there really some calibration values transmitted? Can you record the data with wireshark? All I got, were the instructions to turn on and off the LED, as I uploaded here: https://github.com/AndunHH/spacemouse/blob/main/Reverse-Engineering-Docs/SpaceNavigator-Connected-Calibrate.pcapng 

>Hi everybody .
I found something about atmega32u4 endpoints that may help :
https://www.avrfreaks.net/s/topic/a5C3l000000UcghEAC/t162507
can anybody check this link ?

I checked the linked page, but can not see the real value. Obviously there may be some endpoints which can not be read and wrote correctly by the atmega32u4 simultaneosly. But I have no clue, if this is the problem we face and how to solve this in our use case. There we would need some other specialist.

But first off all, we need to sniff the values send over USB when a original mouse is connected and the calibration in the menu is performed.

Mr-Falat, 2025-02-07:
Hi.
I can do that guys.
i received a new "SpaceMouse Compact" this week for better observation.
but last night when i connected the promicro to pc i see something intresting: when i swetched between softwares in windows, the RX LED of promicro is blinking ! 
I think the driver send speed and axis direction of device separately for each program!

Mr-Falat, 2025-02-07:
first of all pls compare the information of original Spacemouse and replicated  Arduino device.
I found some differences between those in Device Descriptor !

[Original Device.txt](https://github.com/user-attachments/files/18704938/Original.Device.txt)
[Promicro Device.txt](https://github.com/user-attachments/files/18704937/Promicro.Device.txt)

Mr-Falat, 2025-02-07:
Sorry if I sometimes post irrelevant information.
I am not very familiar with USB. And I am looking for the problem through observation and experiment.

Mr-Falat, 2025-02-07:
The issue is likely related to compatibility with the driver. This problem could be related to one of the following differences:

Device Class and Protocol
Difference: In your device, the Device Class is set to 0xef (Miscellaneous) and the Protocol to 0x01 (Interface Association), while the original device has Device Class set to 0x00 and Protocol to 0x00.
Possible Issue: The 3Dconnexion driver likely expects your device to have the correct Device Class and Protocol like the original device. Changing these values could cause the driver to fail to recognize or communicate with your device.
Solution: Changing Device Class to 0x00 and Protocol to 0x00 may solve the issue.

USB Structure (Composite Device)
Difference: Your device is recognized as a Composite Device, while the original device is recognized as a standalone HID.
Possible Issue: The driver might only support standalone HID devices and could have trouble with composite devices.
Solution: To resolve this issue, it’s better to configure your device as a standalone HID.

USB Speed (Full Speed vs Low Speed)
Difference: Your device is recognized as Full Speed, while the original device is Low Speed.
Possible Issue: Some drivers may only support Low Speed. This speed difference could prevent proper communication with your device.
Solution: Changing the speed from Full Speed to Low Speed could resolve this issue.

AndunHH, 2025-02-08:
Great to hear, that you have a original space mouse for observation.

It would be very interesting to observe the commands send and received during calibration!

Regarding the different headers: We can not emulate the USB header completly, because we also need to programm our mouse through the usb device. I hope this seems not to be the problem. The most interesting part is inside the HID descriptor. So it would be interesting if you compare the HID descriptor details of our emulated mouse and the original one. 

You find our emulated header here:  https://github.com/AndunHH/spacemouse/blob/cdf87f5dfbf1dcc4b2358b84481013974514dbed/spacemouse-keys/SpaceMouseHID.h#L35

Zeratt, 2025-06-08:
Is there any progress in controlling speed and axis inversion through the original 3Dconnexion software?

Zeratt, 2025-06-08:
Perhaps I will have spacemouse enterprise in the near future. I'm not sure what will help, but I'll try to build the protocol via wireshark.

Zeratt, 2025-06-10:
I built my SpaceMouse on Hall sensors and, at the same time, decided to see what its communication with 3Dconnexion software looks like via Wireshark. I noticed that the driver receives the packet "0401" about once every 5 seconds. " 01 " here is responsible for turning on the LEDs, and "04" does not change in any way. But the most interesting thing is that the same package comes every time the settings in the driver change. It's as if he expects spacemouse to respond to him. Can you tell me, when you looked at Navigator via Wireshark, you didn't see how the original Navigator responds to this packet?

Zeratt, 2025-08-22:
I think I've found a solution to the problem! I got access to SpaceMouse Wireless and studied the differences in its behavior from your project: I found that the original SpaceMouse sends data about movement and rotation constantly, and your project only sends it when the data changes. Literally commenting out a couple of lines of code, I forced DIY SpaceMouse to send data constantly, and everything worked with the original driver from 3Dconnexion!

AndunHH, 2025-08-22:
@Zeratt Interesting!

I explicitly implemented this behavior (to send data only on change) to emulate the old space navigator. I had no access to a spaceMouse wireless.

Did you wireshark the connection or did you found out that the SpaceMouse Wireless is sending all the time? 

Please more infos ;) 

What lines did you exactly comment out, so we can test it ourself.

AndunHH, 2025-08-22:
I tried it... but the only thing I get working on the windows driver are enabling the rotations or translations at all. 
I i click to invert the behavior or change it's strength, nothing happens. Can you confirm, that infact inverting the direction is possible?

To avoid errors in coding: I pushed my branch https://github.com/AndunHH/spacemouse/tree/alwaysSendData here so you can see how i tested it.

Zeratt, 2025-08-22:
I confirm that the axes were turned on, inverted, and sensitivity adjusted. That is, everything worked, I checked.

In the near future, as soon as possible, I will try to repeat the result on your branch.

AndunHH, 2025-08-22:
Maybe relevant: can you please report which windows and which 3dconnexion driver you are using?

Zeratt, 2025-08-24:
I have investigated this case further and there are some observations:

1. For some reason, now the original SpaceMouse does not output data all the time either, but only when rejected. Apparently, then, for some reason, he was standing unevenly, so he was constantly giving out data.
2. Sensitivity management works fully on Windows 11. And on Windows 10, only disabling axes works. Tested on versions 1.1.0 and 2.0.0.

AndunHH, 2025-08-25:
Very interesting behavior... And somewhat strange!

When I reverse engineered the protocol, I used an older space navigator as reference. You can read it here: https://github.com/AndunHH/spacemouse/blob/main/SpaceNavigator.md and I also have the Wireshark sniff in the repo uploaded.

Can you also do those steps one by one an record in Wireshark what is happening? Maybe more small sniffs with comments are useful to know what I'm looking at ;)

Because the space navigator only sent the data as single reports and not combined?!

AndunHH, 2025-08-25:
Ignoe my commit above, where i tried stuff. ;) I need to look at a real sniff: 
please record the following scenarios:
Scenario A: Start
Hit Record in Wireshark, plug the mouse in, wait for 3s disconnect it.
 -> Let's compare it to this HID descriptor here:  https://github.com/AndunHH/spacemouse/blob/main/SpaceNavigator.md#hid-descriptor-of-a-spacemouse-wireless

Scenario B:
Mouse is plugged in, hit record and just slightly press done on the mouse on top of the right side to trigger a rotation and release it again. (may take e.g. 3s) i want to check on byte ordering etc in this report where only one rotation should be reported.

Scenario C:
Mouse is plugged in, hit record and press some of the buttons. (and tell us which ;))

Thanks!!

AndunHH, 2025-08-26:
Thanks for this wonderfull records. I looked into them and updated my code on this branch:
https://github.com/AndunHH/spacemouse/tree/SingleHidReport

I changed:
-  the HID header for report 1 to exactly the SpaceMouseWireless.
- the update Rate is now 16 ms as in your provided data
- all translation and rotation data are now sent together (there was a small bug before)
- I wrote a markdown file explaining my analysis to save it -> SpaceMouseWireless.md in this branch mentioned above

Unfortunately, the settings still do not apply, but we are one step closere I hope.

When I open the viewer and the settings dialog simultaneously and I rotate the part slightly the whole time: When I change the sensitivity on one of the sliders, there is definitily something happening, because the viewer is shortly stock before rotation continues onchanged.

Can you confirm via wireshark that nothing is transmitted to the mouse, when you change some of the sliders? Please record:
- Connect the controller
- Rotate slightly for the whole time
- change sensitivity in the settings dialog (please tell us, if the viewer also stucks for 1s with the original mouse)
- stop rotation
- disconnect mouse

I will try completely emulate the SpaceMouse Wireless by also changing:
- [ ] USB VID and PID, 
- [ ] the descriptor string 
- [ ] other reports in HID descriptor

Zeratt, 2025-08-26:
[sensitivity.zip](https://github.com/user-attachments/files/21995269/sensitivity.zip)

Filter:
usb.src != 1.1.2 && usb.dst != 1.1.2 && usb.src != 1.13.2 && usb.dst != 1.13.2 && usb.src != 1.1.3 && usb.dst != 1.1.3 && usb.src != 1.1.0 && usb.dst != 1.1.0 && usb.src != 1.2.0 && usb.dst != 1.2.0 && usb.src != 1.13.0 && usb.dst != 1.13.0 && usb.src != 1.4.2 && usb.dst != 1.4.2 && usb.src != 1.4.0 && usb.dst != 1.4.0

I confirm that with my DIY mouse, any 3D modelling programs will freeze for 1 second when switching if the sensitivity settings have changed. If the settings have not changed, then there is no freeze.

This behavior has not been observed with the original mouse.

Zeratt, 2025-08-28:
Now, looking through the WireShark logs again, I found that if the original SpaceMouse consistently outputs data with a pause of 16 ms, then the DIY SpaceMouse outputs data unstable. Sometimes it takes 14-15 ms, and sometimes 18-22 ms. Does this have any effect on how the software processes the data? Considering that the software freezes for a short time when changing settings, I can assume that it is trying to process data, but some kind of exception occurs, and the data is bypassed by the software settings. And the timeout of this exception is precisely expressed in this suspension.

Logs showing this instability of data transmission.
[DIY_spacemouse.zip](https://github.com/user-attachments/files/22015813/DIY_spacemouse.zip)

AndunHH, 2025-08-28:
Interesting observation.

Can you check what happens if movements and keys both are transmitted? Do the keys replace the slot of a transmitted movement or do they maybe take a spot in-between after 8 ms?

I expect the later one, because this would fit to the old behavior with separated translation and rotation each every 8ms. After this two reports the third key report was given after additional 8ms.

Please record moving the mouse and then press and release a button for some times.

Zeratt, 2025-08-28:
The original SpaceMouse sends a packet with buttons 16 ms after the packet with movements. that is, it has a constant break between packets of 16 ms, and the button data is simply embedded in this chain of packets.

DIY SpaceMouse just doesn't send packets with buttons while there is movement. I'll assume that the program just stays in the ST_SENDTRANS state, since it only enters ST_SENDKEYS if there are no movements.

[rot_and_buttons.zip](https://github.com/user-attachments/files/22023367/rot_and_buttons.zip)
Filter: usb.src != 1.1.2 && usb.dst != 1.1.2 && usb.src != 1.13.2 && usb.dst != 1.13.2 && usb.src != 1.1.3 && usb.dst != 1.1.3 && usb.src != 1.1.0 && usb.dst != 1.1.0 && usb.src != 1.2.0 && usb.dst != 1.2.0 && usb.src != 1.13.0 && usb.dst != 1.13.0

AndunHH, 2025-08-28:
I looked into the data our mouse does, when the sensitivity is changed: 
The pc request the LED to light and it takes 1s to answer for our DIY mouse. Maybe this is the same delay we see in the application??? I have no clue why the response to toggling the led takes so long. ... maybe this is something how the UsbCore is handling the request?

AndunHH, 2025-08-28:
I also fiddled around a little bit more: https://github.com/AndunHH/spacemouse/compare/main...SingleHidReport

In my last commit ffcbfea341047c171e3d72b8eced5e7bc4687392 I found out, that when we empty the RX buffer always (to know the LED status) the 1s waiting during sensitivity changes is gone.

nevertheless, this doesn't solve our problem. the changes are still not taken into account

AndunHH, 2025-08-29:
I also just updated my driver and the sensitivity and direction changes are now working in our actual exploration branch and in the actual main branch.

The setup that is working:
Win 11 Home, Version 24H2, Build 26100.4946
3DxWare: 10.9.7.709
3DxWinCore: 17.9.7.21845

The not working version:
3DxWare: 10.8.20.3729
3DxWinCore: 17.8.20.20889

I suggest, that installing the newest driver can be expected from the users. Maybe more people can confirm, that it is working on Win11 with the newest driver, even with the master branch or the release.
@voxpop9 @coliss86 @Mr-Falat Could you maybe try if a driver software update solves this problem also on your machines?

Please report the driver version and which version of the space mouse your using. Thanks!

Zeratt, 2025-08-31:
I checked on my systems:

Windows 10 Pro 22H2, Build 19045.6216
3DxWare 10.9.7.709
3DxWinCore 17.9.7.21845
Master branch - not working
Release 2.0.0 - not working

Windows 11 Pro 23H2, Build 22631.5624
3DxWare 10.9.7.709
3DxWinCore 17.9.7.21845
Master branch - working
Release 2.0.0 - working

Zeratt, 2025-08-31:
Actually, from the very beginning, I had a situation that the driver works on Win 11 and does not work on Win 10.
So far, my Win 10 driver has never worked. Can you tell me, did you have any moments when the driver worked on Win 10?

AndunHH, 2025-08-31:
I have only one pc and that switched to win 11 over a year ago. Back then it was not working. 
(but my main machine is a Ubuntu, therefore I'm not using it on windows on a regular basis)

coliss86, 2025-08-31:
I can't help you, my multi axis mouse is broken :'(
Good luck 💪

AndunHH, 2025-09-21:
@Zeratt Do you have anymore ideas what to do or check to get it working on the windows 10 machine?

I have no further ideas and because it's working on a fully updated systems (Win11 with newest driver). And because Win 10 Updates are stopping in october, I'm inclined to say it's good as it is.

Therefore I close the issue.

If you have other ideas, feel free to comment and we can reopen the issue or create another one, with a new clue.