AndunHH/spacemouse #15: Config comment wrong

skorokithakis, 2024-06-15:


AndunHH, 2024-06-15:
The actual implementation works in the way, that e.g. for the translation in x direction -CY and +AY are summed up.
Regarding this picture: 
[image]

Therefore, to calculate transX the second and sixth value are taken. That spoken: The comment should be in the same orientation as the calibration.h definitions.

I will describe this clearer with this updated picture in the readme.

If you finally need another x, y and z orientation for the 3d connexion driver, adjust this via the InvX etc. settings in your config. 

Does this answer your question?

MikiZed, 2024-06-16:
I agree with @skorokithakis, adn I didn't quite understand the reply. 

The axis you have drawn on each joystick in black and blue are the exact opposite to what's in the note in the bottom right 

Connecting like this:
```
// AY, AX, BY, BX, CY, CX, DY, DX
#define PINLIST { A1, A0, A3, A2, A7, A6, A9, A8 }
```
and looking at the serial monitor each joystick outputs the correct axis just with the wrong orientation (but that's easily fixable in the config or in 3d connection software so no big deal)

skorokithakis, 2024-06-16:
I didn't quite understand the reply either. Even before going into the math, the PINLIST variable is used to define the connections to each joystick. The given comment is simply wrong, as I didn't connect AX to A1, but to A0, no matter what happens with the math downstream.

AndunHH, 2024-06-16:
Hm... First of all: You a right about the picture. The legend in the corner is useless, because it is wrong.
Here is my next try to explain:
[image]

Let's try to get through this in one example and find our misunderstanding:

We calculate the translational movement of the space mouse to the right. (label with the big red X, called "TRANSX" in the source code).
Therefore we need to calculate the horizontal movement of the joysticks in front of us (+A) and at the back side reversed (-C):
1. We need the joystick value from those two axis.
2. I named this horizontal axis of the joystick "y".
3. The applied math is ``TRANSX = -CY + AY```
4. The position of the axes in the PINLIST is defined with constants in calibration.h:
```
...
#define AY 1
... 
#define CY 5
```
5. The horizontal axis ("y") of the joystick in front of you ("A") shall be on the index no. 1 (i.e. the second element) in PINLIST.

Therefore my conclusion is, that the comment at PINLIST must be the same as the numbering in calibration.h for the positions.

Is the problem maybe, that I chose the naming of the axis at the joysticks with y = horizontal different, than your joysticks are labelled? 

I cross-checked with the original file ```joy4_tt_v2.ino``` by [teaching tech](https://www.printables.com/de/model/864950-open-source-spacemouse-space-mushroom-remix/files), which has the same comment as my PINLIST.
```
int PINLIST[8] = { // The positions of the reads
  A1, // X-axis A
  A0, // Y-axis A
  A3, // X-axis B
  A2, // Y-axis B
  A7, // X-axis C
  A6, // Y-axis C
  A9, // X-axis D
  A8  // Y-axis D
};
```

AndunHH, 2024-06-16:
I also updated the config_sample.h with more text to clarify this. Is this clear now? I'm glad somebody is trying to use this code and finds the inconsitencies.

skorokithakis, 2024-06-16:
Ahh, yes, I see now. Yes, my joysticks have "X" and "Y" pins labelled on them, yours don't? That's what I was referring to, because, with your comment, I have to connect the pin that says "A X" to the label that says "Y" on joystick A.

skorokithakis, 2024-06-16:
I have this joystick, you can see the labels here:

[image]

AndunHH, 2024-06-16:
I glad, we found the issue.
My joysticks are (un?)fortunately labeled otherwise .... 
I have to connect the "x" on the joysticks to AX in the PINLIST. 

It depends on the manufacturer of the PCB how he defines the x and y, because it is so easily intermixed but fortunately fixable in software.

skorokithakis, 2024-06-16:
Hm yeah, that's true. These joysticks are recommended by TeachingTech, so I think lots of people will have them. Maybe you can add a comment saying something like "If you have the joystick TeachingTech recommended, the pins labelled X and Y on the joystick are swapped here", or something similar, to avoid people getting confused?

AndunHH, 2024-06-16:
Yes, I added this as a hint and I think we have sorted this puzzle! ;)

closed 2024-06-16 by AndunHH