AndunHH/spacemouse #16: "No such file or directory" regarding a link to include "config.h"

DougJoseph, 2024-06-15:
I am attempting to compile and upload this software, but the Arduino IDE keeps saying "No such file or directory" regarding a link to include "config.h" — so something I've done apparently has me down a path that's different from others.

Would anyone mind sharing their config.h file, or throw me a clue to where I can find one already in my system?

I'm on a Mac. Thanks!

AndunHH, 2024-06-15:
Hi
You need to rename the config-sample.h to config.h and adapt your pin settings. 
Follow the instructions in the readme file here in github which will instruct you there.

DougJoseph, 2024-06-15:
Ah, thanks!!!

AndunHH, 2024-06-16:
If have chosen this behavior, so I don't upload my personal config.h every time I commit some code.

And vice versa: if you update the code from this repository, your config is not overwritten, just the config-sample.h
(May be smaller updates are than necessary, but that's easier than repeating the whole calibration and pin assignment after an update)

closed 2024-06-16 by AndunHH

DougJoseph, 2024-06-16:
That's a wise approach for sure. Thanks!