AndunHH/spacemouse #19: "pins_arduino.h" not found

katzio, 2024-06-16:
**Not a bug, but this might help someone else (maybe add it to FAQ)**

Compling on WIndows 11, 2.3.2 Arduino IDE, AVR 1.8.6

I follow every step, and on complie got error "pins_arduino.h" not found.

Checked the file - indeed, not existing, even though it is imported in [core](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Arduino.h#L258C11-L258C25)

Its the first time I interact with Arduino and I'm not sure if this was suppose to be auto copied, but it didn't

Copying the config from `variants\leonardo` (in my case, I used the board TeachingTech recommended) to version root (`...avr\1.8.6`) solved it

Thanks for the effort you put in this repo, cool project 👍🏻

AndunHH, 2024-06-17:
Did you load the board support for the sparkfun Boards before? 

I also didn't had this issue, but my working copy was building upon an installation of Arduino IDE 1.x

horvatkm, 2024-06-23:
With Arduino IDE 2.3.2 and Arduino AVR board package 1.8.6 , a few changes in text pasted to boards.txt were needed

#spacemouse.build.variant=promicro
spacemouse.build.variant=leonardo

#spacemouse.bootloader.file=caterina/Caterina-promicro16.hex
spacemouse.bootloader.file=caterina/Caterina-Leonardo.hex

Also SpaceMouse needs to be added to list of boards supported by Arduino AVR platform in package_index.json.

            {
              "name": "SpaceMouse"
            },

AndunHH, 2024-06-23:
Thanks for your suggestion!
I can't verify you suggestion, because I mainly use Ubuntu and the windows machine is already scrambled with an arduino IDE version 1.x. ...

As I read it, we may need proper instructions for those three use-cases of our repo:
1. Windows with Arduino IDE 2.x -> putting your suggestion into a proper step-by-step guide
2. Linux  (and windows?) with Arduino 1.x -> Archive the existing description, because it seems to work
3. The PlatformIO instructions, already in the readme, thanks to @skorokithakis in #23. The user interface in platformIO is more powerfull (=overwhelming for newcomers), but the initial setup is trivial, because there is no board creation necessary. 

Do you agree, that we need a second description or do you think we can omit the description for Arduino IDE 1?

Can you write your suggestions from above as a step-by-step instructions with a complete listing of the board.txt, and where to put it?

horvatkm, 2024-06-23:
IMHO better than describing how to configure development environment would be 
- allow user to configure SpaceMouse behavior from console and save configuration to EEPROM
- to provide compiled binary with description how to flash it

I've no idea why variant and bootloader in your setup is named differently, maybe you are using older Arduino AVR board package 1.8.6 ?

Benoit-LdL, 2024-07-05:
This solution worked for me as well.

AndunHH, 2024-09-10:
> IMHO better than describing how to configure development environment would be 
> - allow user to configure SpaceMouse behavior from console and save configuration to EEPROM

This good suggestion is now part of this feature request:
https://github.com/AndunHH/spacemouse/issues/57 

I close the rest of this issue, as it seems to be described by the wiki.

closed 2024-09-10 by AndunHH