AndunHH/spacemouse #27: Change board.txt template with added tool.serial?

AndunHH, 2024-06-19:
As written in the faq, some people need to
change the board.txt:
spacemouse.upload.tool=avrdude to spacemouse.upload.tool.serial=avrdude to get no error when compiling

Can we include this in the general board.txt instructions? Is there a problem for the older installations?

See
-  https://forum.arduino.cc/t/software-upload-error/1269516/7
- https://gist.github.com/nebhead/c92da8f1a8b476f7c36c032a0ac2592a?permalink_comment_id=5069434#gistcomment-5069434

coliss86, 2024-06-19:
Sure you can add this as a note in the readme of this project in a section : "common issue".

AndunHH, 2024-06-20:
Yes, this was the easy part. 

Someone working with arduino and how don't need this line added should check if we can just add it generally.

closed 2024-06-21 by AndunHH