AndunHH/spacemouse #37: Robust zeroing at start-up and new semi-auto calibration for dead-zone

AndunHH, 2024-06-30:
This is a feature request / to do list for future improvements:

In a [printables discussion](https://www.printables.com/de/model/908684-spacemouse-mini-slim-profile-with-easier-assembly/comments/1888017)  the following idea emerged:

**Robust zeroing at power-up**
When the space mouse is powered, the zero value for the center position is taken from a single `read_analog()`.
Suggestion: 
1. Implement a debug command to reset the zero position without restarting the processor
2. Take values over 500 ms and use the mean as zero position. 
3. Implement some "to way off"-detection?

**Semi-automatic dead-zone calibration**
The user has to set the dead-zone by looking at the values.
Suggestion:
* Implement a new debug command (like debug=20 for min-max) which gets the maximum deviation around the zero values over 15s.

DougJoseph, 2024-06-30:
Great idea. Thumbs up!

coliss86, 2024-06-30:
I like the idea. 
However, if the user do not have button, how are you going to send the reset command? unplugging / replugging the mouse is sufficient.

AndunHH, 2024-06-30:
In my first attempt, I would implement such a calibration routine i a way, that you can call it during calibration and debugging via the serial interface.

During normal use later on, an interface like in the 3dConnexion program would be usefull... (but I have no clue how to trigger it... ) 
Therefore, for normal use, just unplugging and replugging the mouse stays the best solution now.
But if someones finds out how to solv #13 we can also use this approach for this calibration.

Until then, the feature is mainly usefull during calibration and intial debugging.

AndunHH, 2024-07-06:
I'm working on it in the following branch 
https://github.com/AndunHH/spacemouse/tree/RobustZero

I just commited a zeroing function. I will write a dead-zone suggestion function, before starting the pull request.

DougJoseph, 2024-07-06:
Sounding good!

AndunHH, 2024-07-09:
This is merged into main, therefore I close this feature request issue.

closed 2024-07-09 by AndunHH