AndunHH/spacemouse #39: weird movement: zoom first and rotation

coliss86, 2024-06-30:
Hi,

I do not think it a bug in the software, I need help to figure out what is wrong:
In fusion, when moving slowly the axis on RX for example, it first zoom out, and when I rotate more the axis, it then rotate.

Here is my `config.h`:
```
#define STARTDEBUG 0
#define DEADZONE 3  // Recommended to have this as small as possible for V2 to allow smaller knob range of motion.
#define MINVALS {-522, -522, -492, -493, -517, -514, -505, -511};
#define MAXVALS {501, 501, 531, 530, 506, 509, 518, 512};
#define TRANSX_SENSITIVITY 2
#define TRANSY_SENSITIVITY 2
#define POS_TRANSZ_SENSITIVITY 0.5
#define NEG_TRANSZ_SENSITIVITY 5  //I want low sensitiviy for down, therefore a high value.
#define GATE_NEG_TRANSZ 15        // gate value, which negative z movements will be ignored (like an additional deadzone for -z).
#define GATE_ROTX 15              // Value under which rotX values will be forced to zero
#define GATE_ROTY 15              // Value under which roty values will be forced to zero
#define GATE_ROTZ 15              // Value under which rotz values will be forced to zero
#define ROTX_SENSITIVITY 1.5
#define ROTY_SENSITIVITY 1.5
#define ROTZ_SENSITIVITY 2
#define MODFUNC 0
#define INVX 1   // pan left/right  // 3Dc: 0
#define INVY 0   // pan up/down     // 3Dc: 1
#define INVZ 1   // zoom in/out     // 3Dc: 1
#define INVRX 0  // Rotate around X axis (tilt front/back)  // 3Dc: 0
#define INVRY 0  // Rotate around Y axis (tilt left/right)  // 3Dc: 1
#define INVRZ 0  // Rotate around Z axis (twist left/right) // 3Dc: 1
#define SWITCHYZ true //change to true for switching movement
#define NUMKEYS 0
#define NUMHIDKEYS 0
#define NUMKILLKEYS 0
#define ROTARY_AXIS 0
```

here is a debug with `STARTDEBUG 5`
```
AX:0, AY:0, BX:0, BY:0, CX:0, CY:0, DX:0, DY:0, ||TX:0, TY:0, TZ:0, RX:0, RY:0, RZ:0,
AX:0, AY:0, BX:0, BY:0, CX:0, CY:0, DX:0, DY:0, ||TX:0, TY:0, TZ:0, RX:0, RY:0, RZ:0,
AX:0, AY:0, BX:-216, BY:0, CX:0, CY:0, DX:64, DY:0, ||TX:0, TY:0, TZ:-304, RX:0, RY:186, RZ:0,
AX:0, AY:0, BX:-348, BY:0, CX:0, CY:0, DX:98, DY:0, ||TX:0, TY:0, TZ:-350, RX:0, RY:297, RZ:0,
AX:0, AY:0, BX:-351, BY:0, CX:0, CY:0, DX:125, DY:0, ||TX:0, TY:0, TZ:-350, RX:0, RY:317, RZ:0,
AX:0, AY:0, BX:-351, BY:0, CX:0, CY:0, DX:150, DY:0, ||TX:0, TY:0, TZ:-350, RX:0, RY:334, RZ:0,
AX:0, AY:0, BX:-351, BY:0, CX:0, CY:0, DX:150, DY:0, ||TX:0, TY:0, TZ:-350, RX:0, RY:334, RZ:0,
AX:0, AY:0, BX:-351, BY:0, CX:0, CY:0, DX:150, DY:0, ||TX:0, TY:0, TZ:-350, RX:0, RY:334, RZ:0,
AX:0, AY:0, BX:0, BY:0, CX:0, CY:0, DX:2, DY:0, ||TX:0, TY:0, TZ:0, RX:0, RY:0, RZ:0,
AX:0, AY:0, BX:0, BY:0, CX:0, CY:0, DX:0, DY:0, ||TX:0, TY:0, TZ:0, RX:0, RY:0, RZ:0,
```

I first though it was due to my joysticks, I replace them with a new set but it is the same.
It behaves quite the same in all axis.

Any ideas ?

Thanks :-)

AndunHH, 2024-06-30:
Let's go through the math and calculations together. As a reference, here is the picture with the axis and the math:
[image]

I'm examining your 4th line of debug=5:
`AX:0, AY:0, BX:-348, BY:0, CX:0, CY:0, DX:98, DY:0, ||TX:0, TY:0, TZ:-350, RX:0, RY:297, RZ:0`

The values for the joysticks are already centered, deadzone zeroed and mapped from min and max to +/- 350.
Your max for DX is 98. Your min for BX = -348.

You pushed down the right joystick a little bit (DX=98) and pulled up the left joystick very much (BX=-348)

With the calculations (see Readme or in the picture):

```
TRANSZ = -AX -BX -CX -DX
TZ = -0 -(-348) -0 -98 = 250
TZ = mod(TZ) / POS_TRANSZ_SENSITIVITY = 1*(250) / 0.5 = 500
constrain values to +/- 350 -> +350
INVZ = 1 -> TZ = -350.

ROTY = -BX + DX
ROTY = (--348 + 98) / ROTY_SENSITIVITY = 446 / 1.5 = 297
modifier function = 1 -> ROTY = 297
ROTY 
```

Therefore: The math is correct, not a bug, as you wrote. ;)

I don't understand your initial line: Did you want to rotate or translate? What is the expected and what is the undesired movement? 

On my mouse, the handling is much easier, when I just push one side down. For example the right one (DX is positive) and all other values are zero. Then the model starts to rotate and .... (!) when I don't lift the opposite joystick at the same amount (BX negative) the model will also translate, because it recognizes, that the whole mouse is slightly pushed down. Therefore, in general, I reduced this down movement significantly with a high value NEG_TRANSZ_SENSITIVITY and the GATE for neg. Z.

In your example on the other hand: You pulled the left joystick up (BX=-348) and pushed the right one down only slightly (DX=98). Therefore it looks like you need to change the pos_Z sensitivity and maybe introduce a Gate value for this.

Writing and thinking of this: First of all, we should check if the potentiometers in your joysticks are working in the same direction as indicated in the pictures and in my setup:

When you push the joystick down, that value shall be positive. 
When you pull it up, the value shall be negative.
Is this correct? Or are your joysticks wired in the opposite? 
This would lead to the sensitivity and Gate values to be wrongly defaulted.

coliss86, 2024-07-01:
Thanks for your answer.

> I don't understand your initial line: Did you want to rotate or translate? What is the expected and what is the undesired movement?

I want to rotate :-) 

Here is the value when I push the joystick down:
```
AX:350, AY:0, BX:348, BY:0, CX:350, CY:0, DX:350, DY:0, ||TX:0, TY:0, TZ:279, RX:0, RY:0, RZ:0,
```
up:
```
AX:-350, AY:0, BX:-351, BY:0, CX:-350, CY:0, DX:-350, DY:0, ||TX:0, TY:0, TZ:-350, RX:0, RY:0, RZ:0,
```

I'll have a try with `NEG_TRANSZ_SENSITIVITY` and `GATE`, thanks :-)

AndunHH, 2024-07-03:
The positive values for down and negative for up look good. Please report, what you found after tuning the sensitivity and gate.

AndunHH, 2024-07-28:
@coliss86 You may check, if the inversion of single axis (as in #47 and merge #48) helps you.

Or is this issue maybe resolved?

coliss86, 2024-08-03:
Thanks for getting back to me ;)
I didn't take time to do something since this report :/
As I can see from #47, the joysticks used are far better than mine, maybe I'll have to change them again.

euzenlee, 2024-09-12:
The precision of 3D printing seriously affects the feel of our operation, in addition 6DOF is really free, our common operation usually will only move or rotate on a plane, but now this situation can be accidentally misused, it is difficult to achieve precise control. I think this is the reason why this project adds an encoder for rotary operation.
The spring grams of the joystick also affect the operation, it feels like the spring used now is too stiff.

AndunHH, 2025-01-20:
Due to lack of further activities, I believe this issue is resolved? 
Please comment, if this is not the case and we reopen it.

closed 2025-01-20 by AndunHH