AndunHH/spacemouse #47: The Spin and Roll functions in Traniner are reversed

starletzzang, 2024-07-16:
Hello, I am a user who likes space mouse. thank you There is two problem.

[image]

 1. The Spin and Roll functions in Traniner are reversed. How can I do it right?
 2. When I placed the encoder library in the sketch folder and compiled it, an error occurred saying enderwheel.h could not be found, so I modified it to be in the current folder as shown below. Previously, it was in quotation marks.
 
[image]

AndunHH, 2024-07-16:
Hi

Thanks for trying the software and exploring your mouse! 

For the direction of movements, you can check the config.h around line 137, where you can invert or even swap some axis.

For the second issue: thanks for bringing this up to attention! This seems to be no problem in plattformio.... Interesting 🤔 

Which commit or version of the code are you using?
Which Arduino IDE are you using? Version 1 or 2?

AndunHH, 2024-07-16:
The current version is already working as suggested by you or not?

Please try and check https://github.com/AndunHH/spacemouse/blob/main/spacemouse-keys%2Fspacemouse-keys.ino#L29

starletzzang, 2024-07-16:
1. Yes, I know how to change direction, but if you spin the stick in the Spin tab, it will roll.
Image attached. ^^

[image]

2. I am using arduino IDE 2.3.2 version, downloaded the encoder module and pasted it into my sketch folder.

starletzzang, 2024-07-16:
Additional explanation:

Switchyz set the value to 1, and in the trainer's setting, the zoom direction was set to forward/backward. If you change the option to up/down here, Spin and Roll will work properly, but the zoom will not go Up/Down, but Forward/Backward.

[image]
[image]

AndunHH, 2024-07-17:
Regarding [the "" or <> in the include of the encoder library](https://github.com/AndunHH/spacemouse/blob/6970a78f760b80c554da18505b6a9d7ec841ae3f/spacemouse-keys/encoderWheel.cpp#L16):
I checked with arduino IDE 2.3.2 and 1.8.19 under Ubuntu 22.04 and it is working with: `#include <Encoder.h>`, if you install the library over the library manager. 
You are right to change the include to `#include "Encoder.h"`, if you just copied the files to your directory.
Because the library manager is the suggested method of installing libraries, I don't consider this as a bug.

Regarding the Spin and Roll Function: [spacemouse-keys.ino#L276](https://github.com/AndunHH/spacemouse/blob/6970a78f760b80c554da18505b6a9d7ec841ae3f/spacemouse-keys/spacemouse-keys.ino#L276)
Your last screenshot, with the changed ROTY and ROTZ in line 278 looks identical to line 281 (when `SWITCHYZ = 0`). Though, I can't see if you also changed the translation values as well. Usually the rotation and translation values need to be switched together, which is done by SWITCHYZ. 

So I can't see a bug here either or you need to provide more informations, as other users didn't report this error by know. 
**If you are happy with your results, we may stop here.** 

If you want to investigate further, we need to following informations:
1. What version of the code did you download? Which release or commit?
2. Let's start with the correct assignment of the joy-sticks: With debug = 5 copy the output of the serial interface and take a screenshot from the trainer app, for the following situations:
3. If you press down the spacemouse, what values are reported? 
4. If you rotate the whole joystick clockwise, what values are reported?
5. If you press down on the joystick only on the right side, what values are reported?

starletzzang, 2024-07-17:
hello? thanks you reply.

What version of the code did you download? Which release or commit?

--> I previously used the[ code here](https://www.printables.com/model/908684-spacemouse-mini-slim-profile-with-easier-assembly).
But I stumbled upon your code and found out that you are the author. Also, my favorite encoder is implemented!

Let's start with the correct assignment of the joy-sticks: With debug = 5 copy the output of the serial interface and take a screenshot from the trainer app, for the following situations:

Initial state (when nothing moves):   (debug value=5)
[image]

If you press down the spacemouse, what values are reported?
[image]

If you rotate the whole joystick clockwise, what values are reported?
[image]

If you press down on the joystick only on the right side, what values are reported?
[image]

ok, And I recently changed from an analog joystick to a hall sensor type joystick module. It is very smooth and has good sensitivity and responsiveness.

I set Deadzone to 30.
Deadzone = 30
If you don't set it to 20 or 30, when you move the stick, it will move a little in the other direction as well.

[image]

AndunHH, 2024-07-17:
The Initial state looks good!

Pressing Down: 
AX, BX, CD, DX values are negative.

The calculations expect the values for AX to increase, when the whole mouse is pushed down. Compare with the following picture: [image]

Clock-Wise: Is the code you copied the whole movement of a clockwise rotation? The whole movement is hard to get a clue from, because I don't know in which line the maximum position is reached. Can you copy the values, when the space mouse is twisted clock-wise but hold still in this extreme position?

Pressing down on the right side: This also seems to indicate, the DX value shall be positive, but is negativ.

Because I'm not sure about the rotation, I can't say if the AY, BY, CY, DY values are also given in the opposite direction.

If you have a code, that is working, this is great. I think, I will add an option to invert the logic of the input pins low-level. This would make other tunings easier. Because I included a gate value like `GATE_NEG_TRANSZ` to avoid values, when the mouse is pushed down. With your setup, this will avoid the mouse being pulled up. You can tweak the settings to work good, but this is more effort, because our joysticks don't behave the same.

There is an easier way to copy such numbers, than a picture. Copy it as a text and insert it here within a code block. You create it by typing three ``` marks.

Identified missing feature: 
- [ ] Add possibility to invert AX etc. on low-level basis

AndunHH, 2024-07-17:
I created a branch and added a Pull Request #48.
If you want to try it, you can swap your joystick X values directly on the low level by using 
`INVERTLIST { 1, 0, 1, 0, 1, 0, 1, 0,} `

(As written above, I'm not sure if you maybe need to swap the Y axis as well)

closed 2024-09-10 by AndunHH