AndunHH/spacemouse #51: Upload Error

Vijo0703, 2024-07-23:
After following the steps described in your post, i can Compile the Sketch without any problem, but I can't upload it on to the Pro Micro.

The Error I get is as follows:

"avrdude: ser_open(): can't open device "\\.\COM7": acces denied

Upload failed: Upload Error: exit status 1"

MMaybe somebody can help me ?

AndunHH, 2024-07-23:
Hi

Have you tried the steps described in the readme "Troubleshooting the upload"?

Especially bringing the controller into boot mode for 8s by double resetting.

This may need some retries...

Vijo0703, 2024-07-23:
Hi, 
Yes I have already tried that. 
I've had a similar problem with another project and then it helped to reinstall the driver. Problem is I can't really find how to reinstall the driver for the Pro Micro. 

Thanks for the fast answer

AndunHH, 2024-07-25:
Just to bring in the correct order here some questions or suggestions:
1. You are using windows, probably?
2. Are you using arduino IDE (which version?) or platformIO?
3. You read and followed: [troubleshooting-uploading](https://github.com/AndunHH/spacemouse?tab=readme-ov-file#troubleshooting-uploading)
4. Disconnect the pro micro and check your USB interfaces 
5. After a double reset, look if there is an additional USB port and remember which it is. 
6. Start over again and click on Build and Upload and while the computer is calculating do a double reset. (If you do it in the "correct" order, the compile and start of upload process may take longer than 8s)

If you don't even find the additional USB port in 5, you don't need to try 6.
I can't tell you about the drivers, because I'm mainly using Ubuntu, but it worked out-of-the-box on a win11 machine.

AndunHH, 2024-07-28:
I tried to write down the steps required in more detail here on the wiki page:
https://github.com/AndunHH/spacemouse/wiki/Uploading-firmware-to-the-Pro-Micro

Please feel free to give feedback, if this worked.