AndunHH/spacemouse #54: Someone has ported this to other boards?

FabioM3, 2024-07-31:
Hi, someone knows other boards that have this project ported to? STM32, ESP32, rp2040, etc.?

AndunHH, 2024-08-01:
There is a list of other known space mouse emulations near the bottom of the readme page.

There is one project with an esp32 listed, but it doesn't seem to come very far.... If you get to know such a project feel free to update the list.

Why do you want to use such a controller? Where is the benefit you are looking for? This pro micro is working and can handle the USB hid protocol and can be programmed in plattformio or Arduino ide. What feature are you looking for?

onolox, 2024-08-01:
Simple, I have other boards at hand and no atmega32u4! And this arduino bord is patetic expensive.
But I will buy and use a atmega32u4 for now, maybe if future I try to port to other board.

FabioM3, 2024-08-01:
Simple, I have other boards at hand and no atmega32u4! And this arduino bord is patetic expensive.
But I will buy and use a atmega32u4 for now, maybe if future I try to port to other board.

euzenlee, 2024-09-12:
This project requires 8 analog ports, other chips don't quite meet the requirements, do they?

euzenlee, 2024-09-12:
https://github.com/jfedor2/magellan-spacemouse

FabioM3, 2024-09-12:
It's not so unique combination, and like the project with rp2040 shows, you can add ADCs... BUT good ADCs are needed... rp2040 have poors ADCs.
I wanted to port it to STM32, the perfect candidate for this, cheap, top ACDs, and have clones with USB-C but I have absolutely no time to do it now.

AndunHH, 2025-04-19:
To link stuff together: 
@anishp55 is working on a esp32 platform in #77

joshuata, 2026-02-03:
I've actually started testing TinyUSB with the existing codebase, so I'll keep working on that and hopefully have a PR in a week or two depending on time constraints. I have a handful of other ProMicro compatible boards around as well as both the hall effect and joystick style mice, so I should be able to test changes pretty thoroughly. Very early testing show that the SuperMini should really just be a drop in replacement, and I know from experience that both the bluetooth experience and battery charging support are quite good as it is very popular in the custom keyboard community.

Do you know if anyone has a pcap or any data on how the bluetooth spacemouse communicates with the driver? I've been kind of assuming it would use the same HID protocols over BLE, but I haven't seen any concrete evidence of that.