AndunHH/spacemouse #57: Set settings over serial interface and save them

AndunHH, 2024-09-10:
@Raulpetru created a fork with a python programm to send the sensitivity values over the serial interface. See
https://github.com/raulpetru/SpacemouseProfiles

With this feature request we want the following features:
- implement a serial interface to accept different sensitivity values. (Check if other values are also useful)
- command to output all active settings over the serial to save them on the PC
- possibility to save the values into the eeprom, so the stay saved even with power down
- possibility to set sensitivities for different profiles, which can be toggled with a button or serial command

raulpetru, 2024-09-10:
> 
- [x] 1. implement a serial interface to accept different sensitivity values. (Check if other values are also useful)
- [x] 2. command to output all active settings over the serial to save them on the PC
- [ ] 3. possibility to save the values into the eeprom, so the stay saved even with power down
- [x] 4. possibility to set sensitivities for different profiles, which can be toggled with a button or serial command

1. Right now I commented out the following variables from `config.h` and declared them as float in `spacemouse-keys.ino` so these variables can be changed right now from the user interface:
```
TRANSX_SENSITIVITY
TRANSY_SENSITIVITY
POS_TRANSZ_SENSITIVITY
NEG_TRANSZ_SENSITIVITY
GATE_NEG_TRANSZ
GATE_ROTX
GATE_ROTY
GATE_ROTZ
ROTX_SENSITIVITY
ROTY_SENSITIVITY
ROTZ_SENSITIVITY
```

2. and 4. The app automatically detects on which window the user is focused so the user can modify the variables and "save" them. Saving them actually does 2 things: it will <ins>create</ins> and <ins>apply</ins> a profile which will be saved to a "profiles.json" file that is loaded each time when you open the Spacemouse Profiles app. Switching to different apps will cause the Spacemouse Profiles to switch profiles automatically (if they are available), so the user doesn't need to explicitly toggle a profile change (this is how the 3DConnection software also works, when you switch apps, it will detect and apply a different profile).

3. Right now the app doesn't offer this feature because Spacemouse Profiles app needs to be running in order to change profiles, if you turn it off, the last loaded profile will remain active. If you reboot the spacemouse without opening the Spacemouse Profiles app, it will use the default values written in the firmware.
But it is possible to make "slots" where to save profiles, and trigger them by button (if you want to use the Spacemouse without extra software, or if you want to use it on a machine where you can't install Spacemouse Profiles)

The app is written in Python using the Textualize TUI (Text-based user interface/Terminal user interface) library specifically for Windows, and the app is not that easy to distribute (you have to install python, make a virtual environment, install dependencies, run the app), this can be a limiting factor for not-so-technical users (but I guess most of us are technical since we talk about a DIY mouse). The installing process can also be further automated.
Depending on what we want to add to this app, I might consider on switching to other UI library, a native one that doesn't run in a terminal (which will also allow more customization and also the possibility to minimize the app to system tray).

raulpetru, 2024-09-11:
I forked your repo and implemented the changes more cleaner than in my repo (see: https://github.com/raulpetru/spacemouse/commit/2a6cab550837b98595b1f72fec21abf18a24268b).
I haven't tested this exact changes yet, but should work, I will test them tomorrow.

It's not the best implementation. I want to write a function on Arduino that returns through serial the variables that we want to be included in the UI, then my app will pick this and build the UI. This way I don't have to hardcode the variables in my app, the app will request the variables from Spacemouse and will add them in the UI. This way we can use the same app with different firmwares.

AndunHH, 2024-09-19:
> I forked your repo and implemented the changes more cleaner than in my repo (see: [raulpetru@2a6cab5](https://github.com/raulpetru/spacemouse/commit/2a6cab550837b98595b1f72fec21abf18a24268b)). I haven't tested this exact changes yet, but should work, I will test them tomorrow.

Great! So we can have a look at your approach!

> It's not the best implementation. Also I might have disabled serial debugging since I commented the initial `if (Serial.available())` statement.

We need to find a way to enable both use cases. Now, the input is waiting for only 2 ms and is trying to parse the input as a number. I suggest we add a little input queue, which evaluates the input, when it finds a line feed (new line key \n"). Then: If just a number is entered, we have the actual debug behavior. If e.g. "set transx 15 \n"  ist transmitted the sensitivtiy for translation X is changed. (I suggest that we find shorter variable names...)

> I want to write a function on Arduino that returns through serial the variables that we want to be included in the UI, then my app will pick this and build the UI. This way I don't have to hardcode [...]

Yes, this is very good suggestion!

raulpetru, 2024-11-10:
I'm back with a new application, take a look when you find time ([DIY SpaceMouse Profiles](https://github.com/raulpetru/DIY_SpaceMouse_Profiles)).
We can use this with different firmwares, as long as they implement a few functions to communicate with the UI application.

I can do later a fork (maybe this week or the next one) of your firmware and apply the changes to make it compatible with the UI application. I haven't documented the process, but if you want to do it yourself you can compare the [firmware from my repo](https://github.com/raulpetru/DIY_SpaceMouse_Profiles/blob/master/SpaceMouse%20Firmwares/HE_Spacemouse/HE_Spacemouse.ino) with [the original ](https://github.com/ChromeBee/Hall-Effect-Sensor-CAD-Mouse-Spacemouse/blob/main/HE_Spacemouse.ino) from @ChromeBee to see the changes.

AndunHH, 2024-11-11:
Great to hear from you again!

At the moment I don't have time to code it myself, but I'm looking forward to review and integrate you suggestions!

JarnoBoks, 2025-04-13:
I have a test version running at the moment where I store sensitivities, modfunc and inversion parameters in the Arduino EEPROM. I can change these with commands send through the serial monitor. Is this something you are interested in? I'll have to cleanup the code quite a bit, but after that I can upload a PR that is reviewable?

AndunHH, 2025-04-14:
Yes, this feature is great for new users to calibrate their mouse without recompiling for every parameter change.

Last time, this ticket got stuck. I suggest, that we first focus on your HES merge and than continue here, maybe?

JarnoBoks, 2025-05-23:
Hi, I've been quite busy with this 'learning' project. I think I have a (basic) version ready (see my fork), but it has yet to be tested extensively. My basic version has support for Joystick/Hall sensors, command switches and exclusive mode and EEPROM storage. But the LED functions, nor the Rotary Encoder, nor the Killswitch functionality are implemented. In config_sample.h you can checkout how to use the EEPROM.

My biggest concern now is the (memory) available on the Arduino Pro Micro. My usage of OOP classes, inheritance and virtual functions enlarges the memory footprint of the application quite a bit and I think I have reached the maximum of the Arduino already. 

That's why I'm abandoning this route. I will be focusing the next period to designing and testing a hardware version based on a ESP32S3 zero. I've already added some code/functionality for this processor, but not sure if this will work. Of course you can cherry- pick the parts of the code you want or can reuse, but I don't think a PR will be an option.

ChromeBee, 2025-05-23:
Jarno,
  I am sure you probably know already but just in case you don't. By default the Arduino IDE reserves some space for OTA updates. You can change this default storage utilisation and increase the size of space available for the sketch.

JarnoBoks, 2025-05-23:
I think the OTA only applies to the ESP? Maybe you meant the bootloader? The issue mainly is the SRAM availability. I'm using a lot of virtual function and they take too much SRAM. Will look this weekend into the ESP and maybe design a PCB for that one. Keep you posted.

ChromeBee, 2025-05-23:
No, the OTA memory usage applies to the Pro Micro as well. I had a project that I couldn't upload the sketch to the Pro Micro as it was too large for the available program memory partition. By changing the memory allocation to not reserve 30% for OTA, I was able to reserve more for the program and get it on and running.
Probably a different type of memory to what is needed for variables and objects when running but maybe not. There were different patition options I didn't understand but I selected the first one that didn't include OTA and it worked.

ChromeBee, 2025-05-24:
Sorry, you're correct. Thinking about it now, it was an ESP32C3.. Apologies for any confusion I caused.

AndunHH, 2025-06-30:
@StefanNouza made a great contribution that is nearly ready to be merged in  #82, to save settings to the eeprom.

This post here is all about the external programm to tune the settings or load them based on used programms. But the eeprom saving was missing. He implemented just that! 

It should be easy to add another function to tune the parameters directly from an external programm, without using the parameter menu for humans.