AndunHH/spacemouse #58: Sharing one of my experimental works

euzenlee, 2024-09-14:
I've been following the space mouse project for a while now and have learnt a lot of new things in this repo. Thanks to the author for this interesting and updated knowledge.
I have modelled a few of the projects mentioned in this repo, and due to the accuracy of the 3D printing, I can only say that they are only just usable for functionality, but hardly usable for practical work.
I modified a space mouse with a one-handed joystick, it's not really 6DOF, but it's stable enough to use at work.It has the advantage of being ambidextrous in my left hand. I hope it can give you some inspiration. Use joystick to control x,y axis and a mpu6050 to control rx,ry,rz. In my country, a 6050 module is less than 1 USD, very cheap. It reads values with i2c and is easy to program. There are two buttons on the front, one to control the joystick to switch to z-axis, and the other to control the opening and closing of rx,ry,rz. Since it's not fixed to the desktop, you can only manually turn on the inputs for the rotary action.
I recently found a VL53L0X module, "World smallest Time-of-Flight ranging and gesture detection sensor",I plan to use it for Z-axis calculations.

[image]

AndunHH, 2024-09-19:
Very interesting report, which makes me thinking: 

Is this a commercial joystick, as it looks injection molded. Which?

I assume, you use it with a USB cord in the back and not wireless, right?

How did you modify the software? Do you use the USB HID interface from this repo and "just" calculate the motion differently? There are probably libraries for the gyroscop? Please elaborate on the details ;)

richardwonka, 2024-09-20:
I used these in the early 2000s for large scale 3D projection and modeling. They were quite usable then; I’d be surprised to find they haven’t improved in 20 years?

AndunHH, 2024-10-07:
I also rebuilt the mouse based on the [Ergonomouse](https://www.printables.com/model/973725-ergonomouse-mk-xx-free-version-6dof-controller-kno) file by Jose.  Using the Hall Effect Joysticks makes such a big difference and makes also our mouse in fact usable and not just a stupid proof-of-concept.

Feel free to read my report: https://github.com/AndunHH/spacemouse/wiki/Ergonomouse-Build

AndunHH, 2025-04-19:
Trying to clean up the open issues ;)

Did you publish your experiments somewhere @euzenlee ? Than we could link them on the start page in the "See also" section. 

Please post here and I'll add it to the list. 

In the meantime I'm closing this issue as there is no open feature request or bug in it.

AndunHH, 2025-04-19:
I enabled the Discussion page here in Github, so we can feature such projects: https://github.com/AndunHH/spacemouse/discussions/categories/ideas

closed 2025-04-19 by AndunHH