AndunHH/spacemouse #59: LED does not come on.

starletzzang, 2024-10-06:
LED does not come on.
Connected 5V and VCC of LED, DIN to pin 5, GND to GND. Firmware is latest 0.8.
Did I miss another fix?
Thank you

ps. The LED bar is 5V, DIN, GND and is the one used in the old Orbion mouse.

AndunHH, 2024-10-06:
Do you have a link or description of the led stripe?

Until now, the led support is "just" to enable a simple led on or off.

Your description "DIN" makes me wondering, that you are using a more intelligent led stripe, that needs the easy FastLed library. This is not too much effort, but I can't test it now, because I don't have it in hand.

starletzzang, 2024-10-07:
i was attached LED image.
It's like an LED that receives data from DIN and controls it with PWM to produce each color. It was used in the Orbion mouse.
[image]

AndunHH, 2024-10-07:
The LED support at the moment is only able to pull the pin to GND or high to 5V.

With this very basic implementation only a simple LED can be driven, when connected like this:

[image]

Do you have a model / ref no. for your LED stripe? 
I'm pretty sure we can get it working with the [FastLED](https://www.arduino.cc/reference/en/libraries/fastled/) library, but this is not implemented yet.

What is your expected behavior, when we start to implement it? What should happen where on the LED stripe?

My simplest suggestion:
Turn all LEDs (=bright white), when the 3D connexion driver tells the mouse to turn the led on. But this is really boring... 

As this is not yet part of the repository, I mark this issue as a feature request as it is not a bug.

starletzzang, 2024-10-09:
thanks reply
Just like you said,
For example, the LED can be specified by color, and the color can also be displayed as a gradient. And for the operation, wouldn't it be possible to create an interesting effect where the LED moves in the direction when the mouse joystick is rolled or twisted? (This is a recent gaming mouse effect)

(The content below is unrelated to the LED, but I have to register the article again, so I'm writing it down.
For example, if you press the stick down twice, the map key (e.g. ctrl+d) of the specified config setting will work, so it would be good to add a function like iso view in CAD. (It would be very convenient when designing CAD if you add a user experience to the mouse stick, such as using three fingers to close the window on the iPad)

stonemandave, 2024-11-29:
Anything new with this. I dont know much about coding and im learning. Im williing to help with this if i can. The 3dc software has a calibrate for leds, this would be cool if it worked.

AndunHH, 2024-11-30:
> The 3dc software has a calibrate for leds, this would be cool if it worked.

I already looked in to this "calibration" in the driver / [USB communictation](https://github.com/AndunHH/spacemouse/blob/main/SpaceNavigator.md#hid-report-of-space-navigator). I couldn't find out, what is acutally going on, when this is pressed. When a space navigator is connected, only USB commands are send to let the LEDs blink.

Therefore, the command for LED on/off is already gathered and implemented, see [SpaceMouseHID.updateLEDState()](https://github.com/AndunHH/spacemouse/blob/8fcf07ef0bb21146920afddcbe680da06e36f653/spacemouse-keys/spacemouse-keys.ino#L295). The command to calibrate is not found for a space navigator.

Regarding the initial topic of this issue: I got myself a [neopixel ring](https://www.reichelt.de/entwicklerboards-neopixel-ring-mit-24-ws2812-rgb-leds-debo-led-np24-p235469.html?&nbc=1) and hope to find some time to integrate it into the mouse as a second option despite only a stupid LED.

AndunHH, 2024-12-08:
@starletzzang I started a simple demonstration how to implement such an intelligent LED ring. You can follow my development with suggestions if you like: https://github.com/AndunHH/spacemouse/tree/neoPixelRing

Here is a small preview, what it looks now:
[image]

AndunHH, 2024-12-14:
@starletzzang I finished my work on the led strip for now. 

I would appreciate a test or review on this pull request: https://github.com/AndunHH/spacemouse/pull/65

With that PR we can close this issue.

closed 2024-12-22 by AndunHH

starletzzang, 2024-12-24:
Wow, thank you for your hard work. I just ordered WS2812B 5050 24 chnanil neopixel LED. I want to try your code. I will try it and contact you. Merry Christmas.