AndunHH/spacemouse #61: Why can't  SM_F  trigger?

chenpaner, 2024-10-08:
I made a Spacemouse, but what I needed more was its keys, so I defined 16 keys, but when I added a lot of keys SM_F it never could be triggered, and all other keys could be triggered, but the debug reply could correctly return Key: 15 .

> Note: SM_FIT, SM_T, SM_R, SM_RCW I used the joystick to trigger these 4 buttons instead, so they are all pins wrong, but they all work

```
#define NUMKEYS 16

#define KEYLIST    {  1,   95,  96,3,   97,21, 20, 15, 97 , 97, 10,  2, 14,16, 0,7}

#define NUMHIDKEYS 16

#define SM_MENU     0 // 按键 "Menu"
#define SM_FIT      1 // 按键 "Fit"         ??
#define SM_T        2 // 按键 "Top"
#define SM_R        4 // 按键 "Right"
#define SM_F        9 // 按键 "Front"
#define SM_RCW      8 // 按键 "Roll 90°CW"  ??
#define SM_1        12 // 按键 "1" 
#define SM_2        13 // 按键 "2" 
#define SM_3        14 // 按键 "3" 
#define SM_4        15 // 按键 "4"
#define SM_ESC      22 // 按键 "ESC"
#define SM_ALT      23 // 按键 "ALT"
#define SM_SHFT     24 // 按键 "SHIFT"
#define SM_CTRL     25 // 按键 "CTRL"
#define SM_ROT      26 // 按键 "Rotate"    这个lockrot  ??

#define BUTTONLIST {SM_3, SM_FIT, SM_T, SM_ESC, SM_R, SM_CTRL, SM_1, SM_SHFT, SM_RCW, SM_MENU, SM_ALT, SM_4, SM_MENU, SM_ROT, SM_2,SM_F}         
```

AndunHH, 2024-10-08:
Wow, your board looks amazing!

Let's try to find out, where the bug might be:

As you say, that the debug output is reporting Key 15, the bug is proably not in the pin read out (and the KEYLIST) itself.

Possible problem a:
Have you tried to swap elements in the BUTTONLIST? Is the problem always the SM_F which is not recognized by your PC?

Possible problem b:
Or is it always a problem with the last entry in the BUTTONLIST? 

Possible problem c:
Have you tried to reduce the list to e.g. 6 Keys? Then please try to test if SM_F is working here. Then we would have a problem with the length of the list...

In either case please provide some **additional information:**
1. Which version of the code are you using? 
2. How are you testing the keys? On a windows PC with 3d connexion driver probably? 
3. In which programm do you expect SM_F to take an effect? 

Possible problem d: 
You misconfigured the 3d connexion driver software and the button is sent to the PC but not considered correctly.

AndunHH, 2024-10-08:
I just tested the SM_F key in general on a windows 11 machine with 3d connexion driver and Onshape. In general it is working:
[image]

chenpaner, 2024-10-09:
A:
The last one is unrecognizable only if it is SM_F, and unrecognizable no matter where SM_F are placed
B:
The last one is recognized by other buttons.
C:
If listed to e.g. 6 Keys, SM_F can work

I am using the latest version of the 3d connexion driver and your latest code version. I have tested in Rhino, Blender, and Creo and failed to trigger correctly SM_F

I'm pretty sure it's the length of the list, and I defined several other keys in the spacemouse-keys.ino file outside the list, and the pins are repeated, but they all work. In fact, this problem is difficult to describe, because I have modified a lot on the basis of your code. This problem may be fixed by myself.

I added a side of the code myself, use Ctrl + 1/2/3/4 to switch the 4 joystick speeds, because I found that the default rotation speed is different in different software (such as Rhino and Creo).

spacemouse-keys.ino
```
#define BT_CTRL   21
#define BT_1        20 // 按键 "1"
#define BT_2        0  // 按键 "2"
#define BT_3        1  // 按键 "3"
#define BT_4        2  // 按键 "4"

// 动态旋转速度缩放
float rotspeed = 1.0f;  // 初始化 rotspeed 为 1.0
int buttonStates = 0; // 用于记录按钮状态
int lastButtonStates = 0; // 用于记录上一次按钮状态
float rotspeedValues[] = {0.5, 1, 3, 6}; // 定义 rotspeed 的各个状态值,4个挡位

void setup() {
#if NUMKEYS > 0
  setupKeys();
#endif
  Serial.begin(250000);
  delay(100);
  Serial.setTimeout(2);  // 串行接口将查找新的调试值，并且只会等待 2ms

  busyZeroing(centerPoints, 500, false);
  
  pinMode(BT_CTRL,  INPUT_PULLUP);
  pinMode(BT_1,     INPUT_PULLUP);
  pinMode(BT_2,     INPUT_PULLUP);
  pinMode(BT_3,     INPUT_PULLUP);
  pinMode(BT_4,     INPUT_PULLUP);
  
#if ROTARY_AXIS > 0
  initEncoderWheel();
#endif
  pinMode(LEDpin, OUTPUT);
}

void loop() {

  int ctrlState = digitalRead(BT_CTRL);
  int oneState = digitalRead(BT_1);
  int twoState = digitalRead(BT_2);
  int threeState = digitalRead(BT_3);
  int fourState = digitalRead(BT_4);

  int currentState = 0;
  if (!ctrlState) currentState |= (1 << 0);
  if (!oneState)   currentState |= (1 << 1);
  if (!twoState)   currentState |= (1 << 2);
  if (!threeState) currentState |= (1 << 3);
  if (!fourState)  currentState |= (1 << 4);

  if (currentState != lastButtonStates) {
    lastButtonStates = currentState;

    // 根据组合键设置 rotspeed
    if (currentState & (1 << 0)) { // 检查 BT_CTRL 是否按下
      if (currentState & (1 << 1)) rotspeed = rotspeedValues[0];
      else if (currentState & (1 << 2)) rotspeed = rotspeedValues[1];
      else if (currentState & (1 << 3)) rotspeed = rotspeedValues[2];
      else if (currentState & (1 << 4)) rotspeed = rotspeedValues[3];

      Serial.print("rotspeed set to: ");
      Serial.println(rotspeed);  // 打印 rotspeed 的值
    }
  }
  // 更新当前状态
  buttonStates = currentState;

  // transY   ADD  / rotspeed
  velocity[TRANSY] = (centered[DX]) / ((float)TRANSY_SENSITIVITY) / rotspeed;
  velocity[TRANSY] = modifierFunction(velocity[TRANSY]); 
  ......
```

chenpaner, 2024-10-09:
#define BUTTONLIST {SM_3, SM_FIT, SM_T, SM_ESC, SM_R, SM_CTRL, SM_1, SM_SHFT, SM_RCW, SM_MENU, SM_ALT, SM_4, SM_MENU, SM_ROT, SM_2,SM_F}
All these keys are output in debug8:
bitnumber: 14 -> keyData[1] = 0x40
bitnumber: 1 -> keyData[0] = 0x2
bitnumber: 2 -> keyData[0] = 0x4
bitnumber: 22 -> keyData[2] = 0x40
bitnumber: 4 -> keyData[0] = 0x10C
bitnumber: 25 -> keyData[3] = 0x2
bitnumber: 12 -> keyData[1] = 0x10
bitnumber: 24 -> keyData[3] = 0x1
bitnumber: 8 -> keyData[1] = 0x1
bitnumber: 0 -> keyData[0] = 0x1
bitnumber: 23 -> keyData[2] = 0x80
bitnumber: 15 -> keyData[1] = 0x80
bitnumber: 0 -> keyData[0] = 0x1
bitnumber: 26 -> keyData[3] = 0x4
bitnumber: 9 -> keyData[1] = 0x2                  SM_F   can`t work!

this is  SM_F   can work output in debug8:
bitnumber: 5 -> keyData[0] = 0x20

AndunHH, 2024-10-09:
> bitnumber: 9 -> keyData[1] = 0x2 SM_F can`t work!

I guess this line corresponds to SM_F when you define 16 Buttons, right?

Here the bitnumber is 9 ... why? SM_F and therefore the BUTTONLIST shall get you bitnumber 5, which is also in your last line.

Can you copy your config.h in the working and not working case here? There must be something wrong with the #define BUTTONLIST and the actual assignment in SpaceMouseHID.h in line 142
`    uint8_t bitNumber[NUMHIDKEYS] = BUTTONLIST;`

Maybe you can add a for loop and iterate through bitNumber and print it on the console. Than you can find out, when the bitnumber is changing from 5 to 9 ??

You did adjust 'NUMHIDKEYS' correctly?

chenpaner, 2024-10-10:
config.h

```
// AX, AY, BX, BY, CX, CY, DX, DY
#define PINLIST \
  { A1, A0, A3, A2, A7, A6, A9, A8 }
  
#define INVERTLIST \
  { 0, 0,  0,  0,  0,  0,  0,  0}
// AX, AY, BX, BY, CX, CY, DX, DY

#define STARTDEBUG 8  // 也可以在程序运行时通过串口设置！

#define DEADZONE 18  // 推荐尽可能小以允许全范围运动。

// 插入测量值如下：{AX,AY,BX,BY,CX,CY,DX,DY}。
#define MINVALS { -512, -512, -512, -512, -512, -512, -512, -512 }
#define MAXVALS { +512, +512, +512, +512, +512, +512, +512, +512 }

#define TRANSX_SENSITIVITY 5
#define TRANSY_SENSITIVITY 5
#define POS_TRANSZ_SENSITIVITY 0.5
#define NEG_TRANSZ_SENSITIVITY 5  //z轴移动除以该值// 我希望向下时灵敏度低，因此使用高值。
#define GATE_NEG_TRANSZ 15        //// 负Z运动将被忽略的阈值（像额外的死区）。 gate value, which negative z movements will be ignored (like an additional deadzone for -z).
#define GATE_ROTX 1             // rotX值小于该值时将被强制为零
#define GATE_ROTY 1              // roty值小于该值时将被强制为零
#define GATE_ROTZ 1             // rotz值小于该值时将被强制为零

#define ROTX_SENSITIVITY 5
#define ROTY_SENSITIVITY 5 //缩放值除以该值
#define ROTZ_SENSITIVITY 5

#define MODFUNC 1

#define INVX 0   // 左右平移  // 3Dc: 0
#define INVY 0   // 上下平移  // 3Dc: 1
#define INVZ 0   // 缩放  // 3Dc: 1
#define INVRX 0  // 绕X轴旋转（前后倾斜）  // 3Dc: 0
#define INVRY 0  // 绕Y轴旋转（左右倾斜）  // 3Dc: 1
#define INVRZ 0  // 绕Z轴旋转（左右扭转）  // 3Dc: 1

// 总共有多少个按键？（0=无按键，禁用此功能）
#define NUMKEYS 16

#define KEYLIST    {   1,     95,   96,      3,   97,      21,   20,     15,      97 ,      97,     10,    2,      14,     16,  0,7}

#define NUMHIDKEYS 16

#define SM_MENU     0 // 按键 "Menu"
#define SM_FIT      1 // 按键 "Fit"         ??
#define SM_T        2 // 按键 "Top"
#define SM_R        4 // 按键 "Right"
#define SM_F        9 // 按键 "Front"
#define SM_RCW      8 // 按键 "Roll 90°CW"  ??
#define SM_1        12 // 按键 "1" 
#define SM_2        13 // 按键 "2" 
#define SM_3        14 // 按键 "3" 
#define SM_4        15 // 按键 "4"
#define SM_ESC      22 // 按键 "ESC"
#define SM_ALT      23 // 按键 "ALT"
#define SM_SHFT     24 // 按键 "SHIFT"
#define SM_CTRL     25 // 按键 "CTRL"
#define SM_ROT      26 // 按键 "Rotate"    这个lockrot  ??

#define BUTTONLIST {SM_3, SM_FIT, SM_T, SM_ESC, SM_R, SM_CTRL, SM_1, SM_SHFT, SM_RCW, SM_MENU, SM_ALT, SM_4, SM_MENU, SM_ROT, SM_2,SM_F}

#define NUMKILLKEYS 0

#define KILLROT 6

[... 528 lines cut ...]
#endif
  
  if (debug == 7) {
    // 更新并报告循环运行频率
    updateFrequencyReport();
  }

  // 通过调用 updateLEDState 检查 LED 状态。
  // 这会清空 USB 输入缓冲区并检查相应的报告。
  #ifdef LEDinvert
   // 如有必要，交换 LED 逻辑
  if(!SpaceMouseHID.updateLEDState()) {
  #else
  if(SpaceMouseHID.updateLEDState()) {
  #endif
    // true -> LED 开 -> 下拉阴极
    digitalWrite(LEDpin, LOW);   // 打开 LED
  }
  else {
    // false -> LED 关 -> 上拉阴极 
    digitalWrite(LEDpin, HIGH);   // 关闭 LED
  }

}
```

AndunHH, 2024-10-10:
I think you have a bug in the definition of the bit number: it should be

`#define SM_F        5 // Key "Front"`

But you have 9 in there... Please check it

chenpaner, 2024-10-10:
Thank you very much! It's really the bug that's causing it

closed 2024-10-10 by chenpaner