AndunHH/spacemouse #62: Is it possible to operate the short key when I click the encoder switch?

starletzzang, 2024-10-18:
How are you? I have one more question.

I installed the encoder and the zoom in/out works normally.

I connected the encoder switch to pin 7.

Is it possible to operate the short key when I click the encoder switch?

AndunHH, 2024-10-18:
I can't see the picture of the schematic. Maybe you need to wait a few seconds, before clicking send.

What due you mean with the "short key"? 
I tested the encoder switch successfully just like a regular key. The controller doesn't need to know if the switch is physically at the encoder or not.

starletzzang, 2024-10-18:
> I can't see the picture of the schematic. Maybe you need to wait a few seconds, before clicking send.
> 
> What due you mean with the "short key"? I tested the encoder switch successfully just like a regular key. The controller doesn't need to know if the switch is physically at the encoder or not.
[image]

AndunHH, 2024-10-20:
Ah, yes. Now I know which button you mean. 

Basically you can configure it as any button you want. 

But are you sure, that the button of the encoder can be pressed? [When I build the knob by Jose](https://github.com/AndunHH/spacemouse/wiki/Ergonomouse-Build#assembling-the-encoder) the encoder was mechanically blocked by using the bearing.

Maybe you should check with a multimeter what happens, when you press the button. Is it connected to "GND" or to "+"? 

The other/usual keys are connecting the signal to GND when pressed.

starletzzang, 2024-10-21:
I missed one application
I got a hint from your answer ""Basically you can configure it as any button you want.""
I connected the encoder SW pin to the existing button and it worked like magic!
And I think I forgot that you can configure multiple buttons and pins in config.h!
Also I temporarily removed the bearing because of the click.
Thank you

closed 2024-10-21 by AndunHH