AndunHH/spacemouse #63: Converter for older devices

dewi-ny-je, 2024-11-08:
Not a bug.

Since you emulate a Spacemouse wireless cabled, what about making also a pass-through converter to connect older devices like Spacepilot and have them appear as Spacemouse?

Those old devices work just fine but are not supported by recent driver versions.

AndunHH, 2024-11-08:
Yeah, this is a nice idea! 

What's missing is the software part to read the data from those old devices.  Or are there already library or code snippets available?

dewi-ny-je, 2024-11-08:
> Yeah, this is a nice idea!
> 
> What's missing is the software part to read the data from those old devices. Or are there already library or code snippets available?

This one maybe? There are links to code snippets 
https://learn.adafruit.com/adafruit-feather-rp2040-with-usb-type-a-host/overview

AndunHH, 2024-11-10:
This board may act as a USB host, yes. 

But I think the bigger problem is, that someone needs to write a device which acts like an old computer, that knows how to talk to this old devices you are speaking off. You need to talk the old protocolls. I was asking for code snippets for this purpuse-
Then someone can write such a translator and use this code here to emulate a modern space mouse.

I actually don't have the need for such a converter, because I build this DIY mouse, someone else could take up your suggestion.

closed 2024-11-10 by dewi-ny-je