AndunHH/spacemouse #64: Keys-only pad

dewi-ny-je, 2024-11-14:
I gave a Spacemouse Compact (so, knob only) as present and I wonder if it would be possible to couple it to a keypad programmed with your firmware to add shortcut keys to it.

This would be interesting because the customisations possible with the 3dconnexion drivers are more advanced than those possible with simple keyboard shortcuts.

AndunHH, 2024-11-14:
I like your idea. Unfortunately our mouse can not sneak into the mouse you already have.  The computer will always detect two devices. Your original mouse and our emulation.

But maybe this is no problem and you can use one mouse for clicking and the other one for pointing? I'm not sure, if 3Dconnexion took this into account or did something against it. ... a quick search led me here: https://forum.3dconnexion.com/viewtopic.php?t=5484 
The discussion is from 2012... so maybe this is worth another look? But I couldn't find anything about it.
When I'm in the company I will try, what happens with two mouse. ;)

As you mentioned, using just a regular keyboard is limited.

One solution is like your idea before: Someone needs to write a software to read the USB signal from the first mouse, before it goes to the PC and change it (=add the buttons). But our software is not able to this.

dewi-ny-je, 2024-11-14:
I connected today a SpaceMouse Compact to my computer while keeping the SpacePilot connected and both worked at the same time, I could grab either knob and start moving objects.
I think that a macro pad using your Arduino code would extend a knob-only device (either genuine, or DIY).

I'm not sure however which macro pads could be wired to use your arduino code: when I search for macro pads I get products which appear as keyboard already. And Arduino keypads don't have good keys

AndunHH, 2024-11-24:
Sry for the delay. I also tried connection our emulated "Spacemouse Pro Wireless" and an actual, real "SpaceNavigator" 

As shown here in the screenshot, the original driver is able to handle both mice simultaneosly, as you already found out.
[image]

So it is in fact very easy to "just" use the buttons to extend a simple spacemouse with a macro pad. 

The actual code is written to run on a atmega32u4 as found on a Pro Micro. It should be possible to convert this to other controllers, but this has not yet been done. See also https://github.com/AndunHH/spacemouse/issues/54 

Sry, I have no link ready to a shop to buy a good quality keyboard with a bare processor meeting our requirements. As of now, this project relies on printing and making...

dewi-ny-je, 2024-11-24:
Probably if you mention this finding and the possibility of using your code just for the macro pad, to extend an existing space mouse or any other supported devices, other people will think about the implementation.
If it's left here in a ticket few readers will find out.

Feel free to close this ticket once the description in the main page is modified to describe the application of the code to a bare macro pad.

AndunHH, 2024-11-24:
thanks for the good hint!

closed 2024-11-24 by AndunHH

AndunHH, 2024-12-13:
> I connected today a SpaceMouse Compact to my computer while keeping the SpacePilot connected and both worked at the same time, I could grab either knob and start moving objects.
> I think that a macro pad using your Arduino code would extend a knob-only device (either genuine, or DIY).
> 
> I'm not sure however which macro pads could be wired to use your arduino code: when I search for macro pads I get products which appear as keyboard already. And Arduino keypads don't have good keys.
> 
> As it's clear, I am excluding designing and 3D printing myself one.

I just read about a crazy open source approach of a keyboard with OLEDs in every key and thought about this ticket. Maybe someone wants to adapt their approach with rasp 2040 to our use case and register this massive keyboard as a space mouse 🐭 

https://www.crowdsupply.com/polykybd/polykybd

AndunHH, 2024-12-13:
Just another finding there: 
This macropad is driven by an atmega32u4 and should be possibly converted to emulate a space mouse as suggested here:

https://www.crowdsupply.com/anavi-technology/anavi-macro-pad-8 

Maybe someone wants to try and convert the source code for it.

The keys seem to be good cherry keys.