AndunHH/spacemouse #66: Newbe Ergonomouse trouble

JoSaetre, 2024-12-14:
I got this fault code mostlikely a newbe mistake (first Arduino programing)

****\Downloads\Ergonomousen\Downloads\Ergonomouse\spacemouse-keys\config.h:37:21: error: 'A7' was not declared in this scope
   { A1, A0, A3, A2, A7, A6, A9, A8 }
                     ^

Compilation error: 'A7' was not declared in this scope

AndunHH, 2024-12-14:
This is a very strange. It seems your code doesn't know that the controller has more than six analog inputs.

Here are some suggestions:
a) Which version did you download? Release 0.8.0?
b) Did you tried to just rename config_sample.h to config.h und still get this error?

c) Did you finished editing the boards.txt file succesfully? We described this here:
https://github.com/AndunHH/spacemouse/wiki/Creating-a-custom-board-for-Arduino-IDE
Hopefully, this is helpful. Otherwise let us know and we describe this better in the wiki.

You should select Spacemouse in the boards selection in the arduino IDE, when it's successfull. Let's check this with a screenshot.

d) Due to the fact, that this is so complicated with the Arduino IDE, I suggest that you give VScode / PlatformIO a try. Changing the board definition is not needed there, after the installation you can "just" open our project and the rest is done automatically. See https://github.com/AndunHH/spacemouse/wiki/Creating-a-custom-board-for-Arduino-IDE.

JoSaetre, 2024-12-14:
a) newest downloaded today
b) yes i did 
c) could not find the file on my computer (probably user error)
d) going with this option

AndunHH, 2024-12-14:
Yes, please try platformIO.
 
The link in my previous comment was wrong, but you'll find the link to platformIO on the readme easily.

If something comes up, feel free to ask.

JoSaetre, 2024-12-15:
I have tried to use Platformio.
This is what I have done: Copied the text of config.h to src-> main.cpp
The other files .cpp and .h I have added to the include folder. and Im down to one fault.

C:..........AppData\Local\Temp\ccNI5dgs.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x86): undefined reference to `setup'
<artificial>:(.text.startup+0x8e): undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
[image]

*** [.pio\build\pro16MHzatmega328\firmware.elf] Error 1

AndunHH, 2024-12-16:
You don't need to copy files around.

Just copy the whole project folder to your PC and open it in vscode. 
This should look like the following picture.
Afterwards the only thing is, that you need to rename config_sample.h to config.h

[image]

The shown platformio.ini is from my actual branch, your's may not show the fastled library, but the rest should be identical.

AndunHH, 2025-05-14:
I assume, the problem is solved? 
We can reopen the ticket, if necessary.

closed 2025-05-14 by AndunHH