AndunHH/spacemouse #71: Rotary Encoder not working

ccomincini, 2025-02-08:
Hi I've printed the full version with knob and wheel and I've a weird issue on the rotary encoder. everytime I move the wheel by 1 click theobject on my cad window start to rotate and continues until i move the encoder by another rotation click. 
 the nly thng I can adjust is on wich axis the object will start rotating. moving the encoder CW or CCW does not make anu difference. the rotation is always in the same direction and it starts and stop click by click.
I've tried three different encoders with the same result.

AndunHH, 2025-02-08:
Which version are you using? The newest version on the main branch or the release? It shouldn't make a big differences. (Just making sure, you are not using a new protoype branch, I just created for another issue)

Did you tried debug=9? What output do you observe? 
First of all, we need to check, that the encoder is correctly counting up or downwards.

Did you check your config file, that the pins are not used more than once? This was a typical problem for me, when testing different functions. Please attach your config file inside a code block.

ccomincini, 2025-02-08:
hi thanks for replying.
here's the config file:
`/* The user specific settings, like pin mappings or special configuration variables and sensitivities are stored in config.h.
Please adjust your settings and save it as --> config.h <-- !
*/

/* Calibration instructions
============================
Follow this file from top to bottom to calibrate your space mouse.
You can find some pictures for the calibration process here:
https://github.com/AndunHH/spacemouse/wiki/Ergonomouse-Build#calibration

Debugging Instructions 
=========================
To activate one of the following debugging modes, you can either:
- Change STARTDEBUG here in the code and compile again or
- Compile and upload your program. Change to the serial monitor and type the number and hit enter to select the debug mode.

Debug Modes:
------------
-1: Debugging off. Set to this once everything is working.
0:  Nothing...

1:  Output raw joystick values. 0-1023 raw ADC 10-bit values
11: Calibrate / Zero the Spacemouse and get a dead-zone suggestion (This is also done on every startup in the setup())

2:  Output centered joystick values. Values should be approximately -500 to +500, jitter around 0 at idle.
20: semi-automatic min-max calibration. (Replug/reset the mouse, to enable the semi-automatic calibration for a second time.)

3:  Output centered joystick values. Filtered for deadzone. Approximately -350 to +350, locked to zero at idle, modified with a function.

4:  Output translation and rotation values. Approximately -350 to +350 depending on the parameter.
5:  Output debug 4 and 5 side by side for direct cause and effect reference.
6:  Report velocity and keys after possible kill-key feature
7:  Report the frequency of the loop() -> how often is the loop() called in one second?
8:  Report the bits and bytes send as button codes
9:  Report details about the encoder wheel, if ROTARY_AXIS > 0
*/
#define STARTDEBUG 0  // Can also be set over the serial interface, while the program is running!

// Generate a debug line only every DEBUGDELAY ms
#define DEBUGDELAY 100

/* First Calibration: Joystick axis pin assignment
==============================================
Default Assembly when looking from above on top of the space mouse
   back    resulting axis (not from the single joystick)
    C           Y+
    |           .
 B--+--D   X-...Z+...X+
    |           .
    A           Y-
  front

Each joysticks has
- a horizontal axis from left to right = Y
- a vertical axis from top to bottom = X
(This definition of X and Y may not correspond to the print on your joysticks... We will find out which signal is X and Y now.)

1. Try to write down the two axis of every joystick with the corresponding pin numbers you chose. 
(A4 and A5 are not used in the example by TeachingTech).
2. Compile the script, type 1 into the serial interface and hit enter to enable debug output 1.
3. At the joystick in front of you (A), move the joystick from the top -> down to bottom (X) and observe the debug output:
  3.a) AX goes from 0 (=joystick at the top) to 1023 (joystick at bottom) -> Everything is correct.  
  3.b) AX goes from 1023 to 0 -> You need to invert AX, see INVERTLIST below.
  3.c) Another output is showing movement: Swap the pins in the PINLIST. Probably you have to swap the first and second element, as AX and AY may be swapped.

If you have the joystick TeachingTech recommended:
    The pins labelled X and Y on the joystick are NOT the X and Y needed here, but swapped. 
    First joysticks Y: AX and X: AY.
[... 322 lines cut ...]
-------------------------------------------------------
With linux and spacenavd: If the space mouse didn't return to absolutely zero in one axis this axis will still report movement, when another direction is pushed, because only the changed values are emitted as events by the linux kernel.

Despite that, values events are emitted with every report send, even if they didn't changed.

Absolute declaration (default)
------------------------------
Every value is always reporting the absolute position. 
This means, in contrast to relative, an axis that is left alone is reported again as zero.
On the other hand, events are only emitted, if at least some value changes. 
This is not always the case, when the space mouse is held still at a non-zero position.
Solution: Jiggling (may be activated by ADV_HID_JIGGLE)
Every non-zero value is reported as it is and +1 in the next report, repeating with +0 in the next iteration and +1 in the next...
This little extra noise is called "jiggling" and ensures that a value declared as absolute is resent with every report, because is not equal to the last value.
*/

// Switch declaration of values to relative, if the following symbol is defined:
// #define ADV_HID_REL 

// Add Jiggling to the value reported, if the following symbol is defined:
// #define ADV_HID_JIGGLE

`

and when I use debug 9 I can see the values growing and decreasing if i turn right or left the encoder

AndunHH, 2025-02-08:
The config looks fine. (for multi line code blocks you need three ` ` ` without spaces to start and stop, fyi)

How many increments does debug 9 output? 

When i turn my encoder a little bit clockwise until it clicks into place, the encoder value is increased by 4. CCW it is decreasing by 4 digits every click. Is this the same for you?

The main part of zooming is done in [calcEncoderWheel](https://github.com/AndunHH/spacemouse/blob/cdf87f5dfbf1dcc4b2358b84481013974514dbed/spacemouse-keys/encoderWheel.cpp#L36)

The number of `ECHOS` defines how often the event for zooming is repeated. When you do debug=7 you get the frequency of the mouse calculating it's stuff. my results are best, when echos is half of this. That means, that a rotation even is fading of over half a second.

The `SIMSTRENGTH` should be how much you push the mouse to get the movement you want. Because we can't rotate harder... just faster, but thats tricky.

Your ECHOS and SIMSTRENGTH should be fine... I suggest you add some debugging lines behind `zoomIterator++;` to see if the simulated pull on the axis is as expected. You may add `Serial.println(simpull);`. When you turn the encoder one click a value should be printed and decrease to zero within a second. Give it a try...

AndunHH, 2025-02-08:
Small addition: Your config file doesn't show the LED pin. Make sure it is not same as the encoder pins!

ccomincini, 2025-02-08:
> Small addition: Your config file doesn't show the LED pin. Make sure it is not same as the encoder pins (and maybe even deactivated)

I've deleted that rows because i do not use leds

AndunHH, 2025-02-23:
Sry for replying so late!

The content of your configuration file looks fine. (Please try to use the code block delimiters :))

Have you tried adding some of the debug suggestions?

> Your ECHOS and SIMSTRENGTH should be fine... I suggest you add some debugging lines behind zoomIterator++; to see if the simulated pull on the axis is as expected. You may add [...]

closed 2025-06-30 by AndunHH