AndunHH/spacemouse #72: Opinions on new idea

alex-military, 2025-02-22:
hello guys i built the first version of this space mauce but i immediately abandoned it because the jerky or imprecise movement was frustrating even after having performed a correct calibration. but apparently things have changed now with the hall effect joysticks.

so i was thinking but what if instead of using the classic joy we used low profile joysticks? (like nintendo switch to be clear) searching online i found them on aliepress but obviously they have a flat connector. Here i thought that instead of mounting 4 sticks on 4 pcb and wiring everything "by hand" we made a single pcb. sure it would be more complex to manage the soldering but i think it would be a better job. do you think it would be possible? in addition another problem that bothered me on this DIY space mouce was the fact that on X and Y translation and rotation the feedback of the springs acts by only 2 while in other cases you have to act on 4 springs simultaneously and this makes it a bit too unbalanced. I was thinking if it was possible to remove the 4 springs in the sticks and use 8 traction springs more or less exploiting the principle of the original space mouse. that is, that the knob is suspended on the springs and self-centered. I await your opinions and maybe I'll get to work on some 3D models and a pcb.

AndunHH, 2025-02-23:
Thanks for coming up with your ideas!

> but apparently things have changed now with the hall effect joysticks.

The main benefit of the hall effect joysticks are good readings for small movement. The whole movement problemns depends more on the effort you put into smoothing the surface of the balls and the bearing inside the main part. I also use some lubricant there! The cheaper the joysticks the more problems are combined. ;) And the hall effect joysticks are really better than the classical ones.

> Here i thought that instead of mounting 4 sticks on 4 pcb and wiring everything "by hand" we made a single pcb. sure it would be more complex to manage the soldering but i think it would be a better job. do you think it would be possible? 

In general everything is possible ;) How do you acutally imagine it? In the actual configuration we place the joysticks circular around the joystick to detect the movements. Now, the central base is moving along the axis of the joysticks. I we just translate the actual configuration to low profile joysticks, this sliding of the inner part is difficult to realize.

There are other configurations possible e.g. with only three joysticks (see https://www.printables.com/model/353764-space-mushroom-full-6-dofs-space-mouse) but they use the same joysticks mechanics.

Can you draw a sketch of your idea of positioning those joysticks?

> in addition another problem that bothered me on this DIY space mouce was the fact that on X and Y translation and rotation the feedback of the springs acts by only 2 while in other cases you have to [...]

I like the idea! Your separating the mechanical attachment from the position sensors. The attachment you are talking about goes in the direction of this approach: https://www.printables.com/model/940040-cad-mouse-spacemouse-using-hall-effect-sensors
Using our joysticks as a sensor instead of his magnets is an interesting approach.

In general I really like the idea of separating the problems:
1. Mechanical attachment of the inner mouse to the housing
2. detecting the position and calculate the position 
3. Processing the data, combine them with buttons, encoders, LEDs. Send them to the PC.

I try to shape this repository to get a good base especially for the third part. If you decide to change the mechanics (part 1)  you can just the software out-of-the-box. If you change the position of the sensors, we can rewrite the kinematics.cpp part and reuse all the other stuff.

Those are my two cents ;)

alex-military, 2025-02-23:
hi you made it clearer for me. for what I meant with the low profile joysticks and the same use of the joysticks used now (if I can I will insert an image with some example joy.) however in the case of those with hall effect not having a physical connector but a flat cable you could make a single board with the flat connectors and the assembly part of the arduino. in addition you could foresee an expansion to be mounted or not on the board to add more buttons.

the only problem with those joysticks is the cost because they cost much more than the standard ones, but they have the enormous advantage of being much smaller, you could save a lot of space that allows us to add other things. anyway for now I will order the normal hall effect

AndunHH, 2025-03-25:
Sry for the late reply. 

Yes, your picture looks very nice and compact. Have you already acomplished a PCB?

In general, the mouse doesn't need to get too small, because it is really benefitial, when the hand rests on the mouse itself. Otherwise the mouse is just pushed around the desk. (the original 3dConnexion mouses solve this issue with ridicoulous high mass and a rubber matt)

The tricky part is to place the button at a comfortable space, where they are not disturbing the movement or joysticks.

AndunHH, 2025-06-30:
If you come up with a new suggestion feel free to reopen this issue. 

In the meantime I suggest the TPU spring version of the hall-effect-sensors with magnet version as shown here: https://github.com/ChromeBee/Hall-Effect-Sensor-CAD-Mouse-Spacemouse/

They are already supported by our software and work really well.