AndunHH/spacemouse #74: Cannot get the 3Dconnexion software to recognise pan and zoom

Kahera, 2025-03-22:
Hey, I hope it's okay to post issues with configuring here as well - not just code specific stuff. 

I'm having an issue where the pan and rotate-functions don't work in the 3Dconnexion software, or in Fusion. However, when I log (using debug mode 4) I seem to get the correct movements in each direction.  

```
Right: 
TX:-350, TY:0, TZ:0, RX:0, RY:0, RZ:0, K0:0, K1:0

Left:
TX:350, TY:0, TZ:0, RX:0, RY:0, RZ:0, K0:0, K1:0

Forward:
TX:0, TY:350, TZ:0, RX:0, RY:0, RZ:0, K0:0, K1:0

Backward: 
TX:0, TY:-350, TZ:0, RX:0, RY:0, RZ:0, K0:0, K1:0

Up: 
TX:0, TY:0, TZ:-350, RX:0, RY:0, RZ:0, K0:0, K1:0

Down: 
TX:0, TY:0, TZ:350, RX:0, RY:0, RZ:0, K0:0, K1:0

---

Tilted forwards: 
TX:0, TY:0, TZ:0, RX:-350, RY:0, RZ:0, K0:0, K1:0

Tilted back: 
TX:0, TY:0, TZ:0, RX:350, RY:0, RZ:0, K0:0, K1:0

Tilted right: 
TX:0, TY:0, TZ:0, RX:0, RY:-350, RZ:0, K0:0, K1:0

Tilted left: 
TX:0, TY:0, TZ:0, RX:0, RY:350, RZ:0, K0:0, K1:0

---

Rotated right: 
TX:0, TY:0, TZ:0, RX:0, RY:0, RZ:350, K0:0, K1:0

Rotated left: 
TX:0, TY:0, TZ:0, RX:0, RY:0, RZ:-350, K0:0, K1:0
```

All rotations are active in 3Dconnexion:
[image]

List of what works and what doesn't: 
[image]

I've got the following setup in my config.h: 
```cpp
#define PINLIST \
  { A1, A6, A9, A8, A7, A0, A3, A2 }

#define INVERTLIST \
  { 0, 1,  0,  1,  0,  1,  0,  1}
--- 
#define DEADZONE 1
--- 
#define MINVALS {-515, -507, -507, -511, -511, -508, -521, -494}
#define MAXVALS {508, 516, 516, 512, 512, 515, 502, 529}
---
#define TRANSX_SENSITIVITY 1.5
#define TRANSY_SENSITIVITY 1.5
#define POS_TRANSZ_SENSITIVITY 0.75
#define NEG_TRANSZ_SENSITIVITY 4  // I want low sensitivity for down, therefore a high value.
#define GATE_NEG_TRANSZ 15        // gate value, which negative z movements will be ignored (like an additional deadzone for -z).
#define GATE_ROTX 5              // Value under which rotX values will be forced to zero
#define GATE_ROTY 5              // Value under which roty values will be forced to zero
#define GATE_ROTZ 5              // Value under which rotz values will be forced to zero

#define ROTX_SENSITIVITY 1.5
#define ROTY_SENSITIVITY 1.5
#define ROTZ_SENSITIVITY 2

#define MODFUNC 0
---
#define INVX 0   // pan left/right  // 3Dc: 0
#define INVY 1   // pan up/down     // 3Dc: 1
#define INVZ 1   // zoom in/out     // 3Dc: 1
#define INVRX 0  // Rotate around X axis (tilt front/back)  // 3Dc: 0
#define INVRY 0  // Rotate around Y axis (tilt left/right)  // 3Dc: 1
#define INVRZ 1  // Rotate around Z axis (twist left/right) // 3Dc: 1

//Switch Zoom direction with Up/Down Movement
#define SWITCHYZ 0  // change to 1 to switch Y and Z axis

//+ some key stuff, but I don't think that should be relevant
```
Note that I've just cloned the current repo, not gone for the code in the last release (noticed the releases way into configuration - whops!)

Been at it for a few hours now, and I'm a bit stumped about what it could be, so I'd greatly appreciate some help!

AndunHH, 2025-03-22:
That's strange, all of your translation value seem to be zero then. They are relevant for the first movements.

Check with debug 6, if the kill key feature didn't delete the translations, when the mouse is moved. (Because debug 4 is earlier in the calculation of data)

(I should implement a debug number to send the values, which are directly send to the USB device in the future.)

Kahera, 2025-03-22:
Oh yeah, that's absolutely the case - debug 6 doesn't give me any T values (while R values work as in debug 4). Thanks for the pointer! 
That would be the `KILLTRANS` button, right? I think I have something funky with my wiring or something then, because the killtrans button doesn't show clicks the same way the killrot button does. I'll look into that 😊

closed 2025-03-22 by Kahera

Kahera, 2025-03-22:
[image]

🤦

AndunHH, 2025-03-23:
Great to hear, that this hint pointed you in the right direction, so you can  help yourself.