AndunHH/spacemouse #77: basic question for ADC resolution

anishp55, 2025-04-10:
I haven't tackled this code base yet, but i am working with the one from teaching teach and have ported a good chunk to the esp32s3.  one part still do not have a clear understanding of is the ADC resolution, on the s3 its fixed to 12 bits, the results are in the 4k range.  do i need to scale these values down for the algorithms? would bitshifting it by 2 bits be enough for the scaling?

AndunHH, 2025-04-10:
This should be relatively easy in our code here. After reading the data, all is scaled to +/- 350, which is fine enough.

Therefore you just need to adjust the readings in kinematics.cpp

The total range is only coded in the math to invert the reading:

rawReads[i] = 1023 - analogRead(pinList[i]);

In every other part, the system just reads your ADC data and takes the center and maps the values around it.

Therefore you definitely need to change the Max and Min value presets in the config.h. but this is also the second or third step during calibration.

I would really appreciate, if you make your esp32 fork public, so we have another option available.

Have you already checked, how to emulate the USB device there? I have not checked how this is done on an esp32 or if the same libraries will do the trick as here

anishp55, 2025-04-10:
i was working with the teaching tech code, and have esp32s3 device recognized as the spacemouse, but registering any movement, so i did some googling and ended up here.  once i get that code working (mostly because its 99% there), i will tackle this

anishp55, 2025-05-22:
Still working on porting this to the esp32.  I just became employed again so might take a bit longer.  Will put up what I have soon in case anyone else wants to help

AndunHH, 2025-09-21:
Did you finish you port to the esp32?

Please feel free to ask more questions in a new ticket or we reopen this ticket. I will close this for now, because there are no open tickets, as far as I see it.

closed 2025-09-21 by AndunHH