AndunHH/spacemouse #80: About the SpaceMouse Enterprise HID Button report

Muratxx5, 2025-04-19:
Hi, thank you very much for your work. It is very inspiring. Inspired by your codes in my own software, I managed to use an old spacepilot device with the new version 3dxware software. Using the spacemouse pro HID button report data here, I recognize the device as space mouse pro to the computer, but I send the data from the spacepilot device. Since the number of buttons is insufficient, I made adjustments according to the enterprise model. I can get the axis data, but I cannot get the button data in this model. Do you have any HID report information for the Enterprise model buttons?

Muratxx5, 2025-04-19:
I think I wrote it in the wrong place. I couldn't delete it either. I'm sorry.

AndunHH, 2025-04-20:
Hi,
it's ok to have it as an issue. You could also open a Discussion, but it's fine.

I did not had my hands on an enterprise model. 

All the USB HID descriptors I got hold off are documented here:
https://github.com/AndunHH/spacemouse/blob/main/SpaceNavigator.md
(We have a Spacemouse wireless and a Space Navigator there)

I also found some infos in the spacenav project, which is writing like a driver for the 3dconnexion devices. You may want to look into their repo and search for enterprise:
https://github.com/search?q=repo%3AFreeSpacenav%2Fspacenavd%20enterprise&type=code 

The button reports are somewhat scrammbled and therefore the spacenav driver is correcting for this misalignment:
https://github.com/FreeSpacenav/spacenavd/blob/36ec926d6836910eb0c54f988417b019f0ce669d/src/dev.c#L468

If you want to emulate such a strange mouse, you also have to send those weird numbers. ;)

Muratxx5, 2025-04-22:
Thank you for your interest. I have reviewed the topics. When I define the Arduino as an enterprise, I cannot send any button data at all. However, I do not have any issues with axis data. I think the HID report is quite different for the enterprise model. So, I continued by defining the Spacemouse Pro model. I am sending the Alt, ESC, Shift, and Ctrl buttons from my own application to the 3dxware software. After excluding these, I am left with 17 buttons. After defining 15 buttons, 2 buttons remain missing, and I cannot use them. I'll handle it this way from now on. If it was an enterprise model device, we would produce HID reports. :(

AndunHH, 2025-04-22:
Nice to hear, that you found a more or less good work around. 

If someone post a HID report of an enterprise we will give you a message ;)

closed 2025-04-22 by AndunHH