AndunHH/spacemouse #90: Information about Bluetooth?

PaddyCube, 2025-09-27:
Hi 
it seems this is the most advanced firmware for diy spacemouse. I want to build one but as wireless. For this, I will try to convert the Code for nrf52840. It hasn't enough ADC I know, but an ADS1115 will be added.

I'm curious about how to establish BLE connection. Do you know if
1. Spacemouse can be connected to PC without a dongle?
2. If it requires a dongle, if USB protocol is the same than the wired Version?

In cae of 2, I will use another nrf52840 as dongle emulating a wired mouse and define my  own protocol for sending data to dongle.

I saw you recently added some insights to a wireless but connected mouse.

AndunHH, 2025-09-27:
Hi @PaddyCube,
building it wirless is a very interesting idea!

Did you already built a usual, wired version? Converting the usual mouse to a different controller with added I2C-ADCs sounds like a sub-project in itself.  This should be possible. Here some thoughts:
- you need to refactor kinematics.cpp, especially `readAllFromJoystick()` to integrate an i2c sensor
- because we slowed the data transmission down to each 16ms, that should be possible over i2c.
- To check if the new controller has enough memory, check https://github.com/AndunHH/spacemouse/blob/main/testConfig/0_build_report.md where I computed the actual memory consumption with different configs. (of course you could e.g. completely dump the LED library if not needed). The minimal version takes 17 kB of flash and 676 bytes of Ram.  A quick googling brought me to this page https://www.nordicsemi.com/Products/Wireless/Bluetooth-Low-Energy. Check if the ram is sufficient!

The second sub-project: Replacing the USB data transmission (handled in SpaceMouseHID.cpp nowadays) with a bluetooth transfer. Here some thoughts about:
a) directly via bluetooth
- I have no idea, how the 3dx driver is communicating here. sry ;( In [issue 13](https://github.com/AndunHH/spacemouse/issues/13) @zerrat was using a wired version, when we sniffed the communication. I own just an old space navigator and can not help with sniffing bluetooth.
- In general, that can't be that complicated. I just asked chatGPT :)

> Modules/SoCs: Chips like Nordic’s nRF52 series, ESP32, or Microchip RN-42 have built-in Bluetooth HID profiles. You just feed them your HID report descriptor and send/receive reports.
>
> BLE (Bluetooth Low Energy) HID: Newer devices mostly use BLE HID instead of "classic" Bluetooth HID. The concept is the same, but HID reports are transported via GATT services instead of L2CAP channels. This is even closer to USB HID descriptors in practice.

b) through a self-built dongle, so the PC thinks it's a wired version.
- Sound's easy: Just run the actual software with it's  SpaceMouseHID.cpp  stuff on the dongle and sent the data, that you receive somehow.

PaddyCube, 2025-09-27:
Wow, thanks for this fast answer, I wasn't expecting this. Also thanks for highligthing the relevant codes and files.

I think I will do this in iterations. First try to get the stuff running on nRF52 as wired version. Afterwards, I'll try to create my own dongle. This might be the fastest way of getting it up and running.

I'll let you know once I get some results.